KiCad/kicad-symbols #1114: Should USB port GND be renamed to USBGND?

danheidel, 2018-11-02:
I'm running into an issue where the GND pin on the USB C Receptacle USB 2.0 and the USB C Receptacle (There may be other parts with this issue, I haven't done a thorough check) in the Connector library overrides the net label and forces any attached net to be named GND.  This causes issues when trying to add a ferrite bead in series with the ground pin.

I have a more detailed writeup with screencaps here: https://www.reddit.com/r/KiCad/comments/9td841/having_an_issue_with_net_names_a_usb_port_pin/

I seem to have ruled out the part in general, the ferrite bead and the GND pin being a stacked power port as the causes.  I looked through the details of the individual power pins (A1, A12 B1, B12) making up that GND pin but didn't see anything that would make it behave differently than the expected behavior.

poeschlr, 2018-11-02:
No but the hidden pins should not be of type power input. (Renaming does not really fix the underlying issue of these pins being global labels)

poeschlr, 2018-11-02:
All usb symbols seem to use hidden power input pins for power supply pins. Here the screenshot from the pin table for usb_c_receptable

[image]

danheidel, 2018-11-02:
I just did a quick and dirty test where I set A12, B1 and B12 to type unspecified (didn't know what an appropriate category for them should be).  Doing so seemed to fix net renaming issue.

Does setting a pin type to power make it a global label?

poeschlr, 2018-11-02:
Invisible power input pins are global labels. Explained in your reddit post and in more detail here:
- https://forum.kicad.info/t/electrical-type-of-schematic-symbol-pins/9439
- https://electronics.stackexchange.com/questions/392911/kicad-5-what-is-the-significance-of-the-various-gnd-symbols/392977#392977
- https://forum.kicad.info/t/tutorial-how-to-make-a-symbol-from-scratch/13336/7

evanshultz, 2018-11-02:
@danheidel 
I just merged the fix so if you try the newest library the issue should be resolved.

closed 2018-11-02 by evanshultz

danheidel, 2018-11-02:
Awesome, thanks!  That was some fast work.

evanshultz, 2018-11-05:
I think I merged the PR and so it's really my fault. I know http://kicad-pcb.org/libraries/klc/S4.3/ says that stacked power pins should be of type Passive and I must have missed it (though our check script should have caught it and kept me from making this mistake...).

poeschlr, 2018-11-05:
I checked the report for that pull request https://travis-ci.org/KiCad/kicad-symbols/builds/433348780?utm_source=github_status&utm_medium=notification

It found the problems with the pins but did mark the request as ok. I have no clue as to why.

diegoherranz, 2018-11-09:
I have found why the errors were reported but Travis didn't mark it as fail.
I'll submit a pull request to kicad-library-utils as soon as I get some time.

poeschlr, 2018-11-12:
Could you document your findings in an issue over at the utils repo? That way it can be fixed by someone else should you for some reason not find time for it.

diegoherranz, 2018-11-17:
I didn't find the time until now (not even to document the issue). But I've submitted a PR now :)

commit b24c6063 (2018-11-02): Fix el. type of USB_C power pins (Fixes #1114) (#1115)

USB_C has stacked power input pins. The invisible pins of the stack can
not be of type power input as this converts them to global labels.
This commit changes them to type passive.