KiCad/kicad-symbols #1122: Feature request: NC markers in Eeschema should modify net names in Pcbnew

danheidel, 2018-11-04:
Minor issue, but it would be nice if putting an NC flag on a part pin in eeschema would then propagate into pcbnew to modify the net names marked on the part pins there.  E.g.:

https://imgur.com/a/X47mcDc

Here, I've got 3 eeschema pins marked with nc flags, 1, 2 and 10.  While not a huge issue, it would be nice in pcbnew, for those same pins to be marked with an NC prefix, or to simply be named NC rather than Net-(U2-Pad-[1/2/10]).  It would help in keeping track of pins, especially on high pin count parts.

antoniovazquezblanco, 2018-11-05:
Although I agree that it is very useful and a nice feature, I think this should be reported in KiCad bugtracker as we, librarians, cannot solve this issue. Please report at:

https://bugs.launchpad.net/kicad

Thanks!

danheidel, 2018-11-05:
Damnit, I should have been paying more attention to where I was posting this.  I'll post it over there, thanks.

antoniovazquezblanco, 2018-11-05:
Haha, no problem. Thanks for the understanding. I will suscribe to that feature request as I am also interested ;)

closed 2018-11-05 by antoniovazquezblanco