KiCad/kicad-symbols #1126: KLC: S4.4 exception for purely passive components

lorem-ipsum, 2018-11-08:
I currently have an open [PR](https://github.com/KiCad/kicad-symbols/pull/605) for a set of passive RF filters. In discussion with @evanshultz all the pin types are set to passive as the compenent itself is purely passive. Now the library check is complaining about a violation of KLC S4.4 as the `GND` pins are also set to passive, not power input or output.

@evanshultz argument for having them as passive was that it is possible to create a PCB with this components, but without power supply (RF in -> filter -> RF out). If it would be set to power input the ERC would complain about the missing power supply, hence the passive label. Personally I think this argument is correct.

Should this added as an exception to the KLC or at least to the lib checker?

antoniovazquezblanco, 2018-11-08:
I think I am not suited to talk about KLC but here is my opinion about the check script.

I would say that the check tool is there in order to help. Here it is outputting a false negative that could be perfectly ignored by the reviewer. I agree that it would be nicer to make the checker script more accurate but one should be as close to check whatever KLC dictates as possible. Adding a exception can let other errors go unnoticed.

Let's see what other librarians think.

evanshultz, 2018-11-08:
I agree. The check scripts are helpful to avoid missing any major issues during a review but there are lots of corner cases where it's just not easy/viable to get complete coverage or try to capture the intricacies of every single part type in the library.

Similarly, KLC is only a set of guidelines and cannot apply perfectly to each part. Some parts of KLC don't apply and we routinely consider the applicability for each PR. So not only does the check script have "holes" but so does KLC!

It's a great question to ask, and if there is time and motivation I think many improvements could be made to the check scripts, but it's known that there is required human intervention during a review so we leave it up to the humans to consider each PR uniquely and in the right context. If you have ideas, time, and skill to improve KLC or the check scripts there are many great opportunities!

myfreescalewebpage, 2019-03-05:
@lorem-ipsum @evanshultz maybe this issue can be closed now I think ? Thanks for the update!

antoniovazquezblanco, 2019-03-05:
@lorem-ipsum Do you have any idea on how to improve the checking algorithm for this particular case? If not, do you think changing the wording of the output would help with your problem?

Thanks!