KiCad/kicad-symbols #1128: move amphenol magjack to "generic magjack" ?

karlp, 2018-11-08:
Shouldn't the symbol added in https://github.com/KiCad/kicad-symbols/pull/535 be renamed to something generic like "RJ45-MAGJACK-LEDS" or something? along the lines of the all the 8P8C generic variants?

It's currently named the very explicit RJ45_Amphenol_RJMG1BD3B8K1ANR but the symbol is the same for ~any 100meg magjack, just the footprint pin numbering sometimes changes.

I want to add a Hanrun HR911105A footprint/symbol, and the _symbol_ seems appropriately generic, but just needs new aliases made?

I've never worked with this git repo before, so
a) is this a correct understanding and is it desired
b) do I do a commit to rename it to generic, then a commit adding each variant?
How does that relate to the footprint repo? do they have to be paired?

Sincerely,

evanshultz, 2018-11-08:
Yes, I think it could have been a generic symbol.

However, note that the windings inside the part result in a specific pinout. The RJ45 pinout is known, of course, but the winding configuration drives a particular pinout of the symbol. A different transformer winding configuration may result in different pin naming so that should be understood before making any changes that don't consider other part variations we might see in the future. The pin names (not numbers) at the link you sent seem good to me to genericize this, but there's been a lot of discussion on pin naming. I favor introducing pin names instead of only numbers for situations like this, but so far it's unsettled. See https://github.com/KiCad/kicad-symbols/pull/592 for another example.

1. I'd wait for a couple other libraries to weigh in and get consensus on the direction of the symbol first.
2. I would reference this issue, once everything is settled here, and make a PR to rename the existing symbol. Just do it on your local fork and then push the commit. If we've figured everything out correctly (pin names, footprint filter, etc.) you can add a footprint and it will all work out.

karlp, 2018-11-18:
See https://github.com/KiCad/kicad-symbols/pull/1146 contains a commit copying it to generic, (and a second one adding a second device using the symbol)

myfreescalewebpage, 2019-01-15:
Just seen this issue after writing at #1146.
@evanshultz many parts like the Amphenol already in the library have the same pinout. I propose to have a generic symbol, with pad numbers based on the Amphenol example. I will still be possible to propose a variation with another generic name in the future if required.
"RJ45_Magjack_LEDs+Shield" can be a good name to describe this generic symbol according to me.
What do you think ?
Joel