KiCad/kicad-symbols #1164: Manufacturer names not in symbols (but are in footprints)

evanshultz, 2018-11-23:
We do not put manufacturers names in manufacturer-specific symbol names. For a good example of an entire library like this, check out https://github.com/KiCad/kicad-symbols/blob/master/Converter_ACDC.lib.

In contrast, we do do this on footprints and it totally makes sense for footprints. The corresponding footprints to the above symbol library are at https://github.com/KiCad/kicad-footprints/tree/master/Converter_ACDC.pretty.

Getting back to symbols, almost all symbols are unique so there's little value in adding `Microchip_` to the front of all PIC MCU symbol names or other library names which are already vendor-specific. And in some cases it would be annoying, like a ubiquitous opamp (like TL07x) or standard logic symbols which are made by many vendors.

However, there are some cases (or perhaps libraries) where I wonder if it would be helpful. Certain libraries have generic symbols but others are all vendor-specific and still others are a mix. https://github.com/KiCad/kicad-symbols/pull/1149 corrected the vendor name being added where I missed it at https://github.com/KiCad/kicad-symbols/pull/1116. In that case, grouping all the XP Power modules may be helpful?

And I just stumbled upon a symbol which does have the vendor name in it:
[image]

I'm fine with leaving it off, and it doesn't seem to hurt anything though for the AC-DC and DC-DC converters we do add (with good reason) the vendor name in the description because that's useful. I don't want to overburden things or make KLC unnecessarily thorough when fewer common-sense rules are better. Still, I wanted to have the conversation and be on the same page about it.

myfreescalewebpage, 2019-01-24:
Same position, I don't think manufacturer is useful in the symbol, but for specific footprints it is.
When making a symbol review I always ask to remove manufacturer from the name of the symbol when the submitter add it.

Do you have established a list of some symbol to be renamed ? Or maybe we can start a list here ? and book this for v6 ?

Joel

evanshultz, 2019-01-24:
Only the AC-DC and DC-DC symbols sprang to mind, and that's because they have a specific vendor name in the description. Otherwise it might be harder to track down info on the symbol (disregarding the datasheet, of course).

The connector I showed above can have it removed unless there's a reason I'm not seeing to keep it.

fauxpark, 2019-01-24:
For reference, this seems to be all the symbols with manufacturer in the symbol name:

```
Amplifier_Audio:STK4*_Sanyo

Analog_Switch:DG417LDJ_Maxim
Analog_Switch:DG417LDY_Maxim
Analog_Switch:DG417LEUA_Maxim
Analog_Switch:DG418LDJ_Maxim
Analog_Switch:DG418LDY_Maxim
Analog_Switch:DG418LEUA_Maxim
Analog_Switch:DG419LDJ_Maxim
Analog_Switch:DG419LDY_Maxim
Analog_Switch:DG419LEUA_Maxim

Connector:Micro_SD_Card_Det_Hirose_DM3AT
Connector:Microsemi_FlashPro_JTAG-10
Connector:RJ45_Amphenol_RJMG1BD3B8K1ANR
Connector:Samtec_ASP-134602-01
Connector:Wuerth_7499010121A
Connector:Wuerth_7499151120

Diode:Central_Semi_CMKD4448
Diode:Central_Semi_CMKD6001
Diode:Comchip_ACDSV6-4448TI-G
Diode:Comchip_CDSV6-4148-G
Diode:Comchip_CDSV6-4448TI-G
Diode:Panasonic_MA5J002E
Diode:Rohm_UMN1N
Diode:Rohm_UMP11N
Diode:Toshiba_HN1D01FU

Display_Character:EA_T123X-I2C
Display_Graphic:EA_*

Driver_Motor:Pololu_Breakout_A4988
Driver_Motor:Pololu_Breakout_DRV8825

LED:Inolux_IN-PI554FCH
LED:Inolux_IN-PI556FCH
LED:LED_Cree_XHP50_12V
LED:LED_Cree_XHP50_6V
LED:LED_Cree_XHP70_12V
LED:LED_Cree_XHP70_6V

MCU_Module (this could be an exception since most of them are known primarily by marketing name or may not have PN at all)

Relay:FINDER-* (PN without manufacturer prefix seems especially un-descriptive, perhaps why hyphen is used instead of underscore)
Relay:Fujitsu_FTR-F1A
Relay:Fujitsu_FTR-F1C
Relay:TIANBO-HJR-4102-L (no mention of "Tianbo" in datasheet...)
Relay:SANYOU_SRD_Form_A
Relay:SANYOU_SRD_Form_B
Relay:SANYOU_SRD_Form_C

RF_GPS:ublox_SAM-M8Q

Switch:SW_E3_SA3216
Switch:SW_E3_SA3624
Switch:SW_E3_SA6432
Switch:SW_MEC_5G
Switch:SW_MEC_5G_2LED
Switch:SW_MEC_5G_LED
Switch:SW_MMI_Q5-100
Switch:SW_NKK_GW12LJPCF
```

myfreescalewebpage, 2019-01-24:
Agreed for some MCU module, particularly Arduino for example etc. When it's possible, the part name should be used, for example we recently merged VisionSOM module from SomLabs.

ublox SAM M8Q is tyoo be reworked too for consistency with other ublox, I have recently merged several ublox symbols (LEA, MAX and NEO series).

Probably we can think to one PR for each family to split the work, one for Analog Switches, one for Driver Motor etc.

serkanxselcuk, 2020-04-11:
How would you think of following suggestion for v6:
- New field ¨man¨ for manufacturer
- New field ¨mpn¨ for manufacturer product number and is same with symbol name in most cases.
- New field ¨sub¨ for substitute (yes/no)
- Manufacturer in symbol name would not more needed, if they have same pin names. (I don´t know any device with same name but different pin names.)

Because PCB manufacturers ask for those information, if we order them to assembly. Actually I place those in each project as global fields and fill them manually to place them in BOM.
Specific parts, even in resistances are needed in some critical industries like vehicle, space, medical, defense... If it is not necessary for user (i.e. only prototype or hobby), he can ignore those fields.

**For common symbols like resistance:**
man: not filled
mpn: =value
sub: yes
If user wants to use a specific resistance, he can change sub field to no, fill man and mpn fields. mpn and value would not be same in this case.

**For symbols with specific device, but several manufacturers like ULN2003:**
man: filled with all of manufacturers
mpn: =value
sub: no
User can delete manufacturers, which he want to ban.

**For symbols with specific device, only one manufacturer like PIC16F84A:**
man: filled
mpn: =value
sub: no
If user need, he can define right mpn inclusive temperature range and packaging, which will be shown in BOM.

This brings benefits for users who want to outsource assembly. I know, it will be a work for all parts, I just wanted to bring this to discussion.