KiCad/kicad-symbols #1165: ADC1173 symbol in library Analog_ADC.lib is wrong

FrankBuss, 2018-11-24:
Pin AIN is wrong, it is pin 19, not pin 27. There is no pin 27 in a TSSOP24 package.

evanshultz, 2018-11-24:
Hi Frank,

It appears this is the first contribution to KiCad! Thanks for taking the time to contact the librarian team, and sorry that it's because you found a possible error.

You are absolutely right:
[image]

Pin 19 should be named "VIN" according to the datasheet as well.

I've submitted a PR to fix this symbol at https://github.com/KiCad/kicad-symbols/pull/1167. I wanted to get this fix implemented right away so I went ahead and made the PR, but in the future if you find an issue you're welcome to submit a PR yourself, if you wish.

Thanks again!

antoniovazquezblanco, 2018-11-26:
PR merged! Please @FrankBuss check if this solves your issue and feel free to reopen the issue if not.

Thank you!

closed 2018-11-26 by antoniovazquezblanco