KiCad/kicad-symbols #1185: new symbols ?

achiestdragon, 2018-11-28:
not shure how to do this so hope this is ok 

 4 new schematic symbols for adding to the libs , 

4067    16 to 1 analog mux /demux     (to 4xxx lib )
ad7846  16 bit  digital to analog converter ( to analog dac lib )
ad978    16 bit  analog to digital convertor (to analog adc lib )
ca3046  5 transistor array  (to transistor array lib)
icl8038   audio function generator / oscilator  (to analog or audio lib )

cant count thats  5 parts so 4+ one free :)

[libs_1.tar.gz](https://github.com/KiCad/kicad-symbols/files/2625968/libs_1.tar.gz)

evanshultz, 2018-11-28:
The best way to add new symbols or footprints is to submit a PR. You may try the official GitHub Desktop application (https://desktop.github.com/) and following the directions at https://help.github.com/desktop/guides/contributing-to-projects/.

achiestdragon, 2018-11-29:
ok , but i still have not figured out how to add these to the existing libs , let alone how to git clone them, 
 anyway heres the other data etc and images for them , may take me some time to sort out the git stuff 

[image]

4067 16 to 1 analog mux /demux http://www.ti.com/lit/ds/symlink/cd4067b.pdf
ad7846 16 bit digital to analog converter https://www.analog.com/media/en/technical-documentation/data-sheets/ad7846.pdf
ad978 16 bit analog to digital convertor (javascript:openreq('http://pdf.datasheetcatalog.com/datasheet/analogdevices/61476620AD976_A_c.pdf') )
ca3046 5 transistor array (https://www.bucek.name/pdf/ca3046.pdf)
icl8038 audio function generator / oscilator (https://www.intersil.com/content/dam/Intersil/documents/icl8/icl8038.pdf)

achiestdragon, 2018-11-29:
also the 4067 is also   alias   cd4067 and also 74hc4067 , not shure how to do these alias also as there across libs would that have to be individual symbols for there alias's ?

evanshultz, 2018-11-30:
Please split up the PRs by obvious groups. One good way is by the library. You could put all transistors together into one PR, for example. Also be sure to use the standard symbols (the transistors are not the current style in the library).

I'm not sure what you mean about the 4067. There are already two symbols present in the library and the symbol, even if there are slight pin naming differences between vendors, can still be kept instead of adding new symbols. In the PR, you can add other ALIASes. What other library would use 4067?
[image]

myfreescalewebpage, 2019-01-24:
@achiestdragon does the previous response answer your questions ? I haven't seen PR from you, do you still want to propose these symbols ?

Thanks for the update !

Joel

myfreescalewebpage, 2019-03-15:
@evanshultz I propose we can close this issue, seems the author is no more working on the subject. What do you think ?

closed 2019-03-15 by evanshultz