KiCad/kicad-symbols #1213: State of the lib regarding 5.0.2

poeschlr, 2018-12-03:
What is the state regarding the 5.0.2 release? Is there anything still to do? (Otherwise i will tag tomorrow morning before i go to work. So about 06:45 central european winter time.)

---

Report by the "check for design breaking changes" script:

Pins have been moved, renumbered or removed in symbols:
-  '4xxx_IEEE.lib:4067'
- '4xxx_IEEE.lib:4508'
- 'Analog_ADC.lib:ADC1173'
- 'Connector.lib:AudioJack2_Ground_Switch'
- 'Connector.lib:AudioJack3_Ground_Switch'
- 'Connector.lib:XLR3_AudioJack2_Combo_Ground'
- 'Connector.lib:XLR3_AudioJack3_Combo_Ground'
- 'Converter_ACDC.lib:VTX-214-015-103'
- 'Power_Management.lib:BTS40K2-1EJC'
- 'Power_Management.lib:BTS50055-1TMA'
- 'Power_Management.lib:BTS50080-1TEA'

Normal pins ok but NC pins have been moved, renumbered or removed in symbols: 
- 'Interface_Ethernet.lib:KSZ9031RNXCA'

Full report will be uploaded shortly

antoniovazquezblanco, 2018-12-03:
4xxx_IEEE were bugfixes as stated by you. See https://github.com/KiCad/kicad-symbols/pull/1159 for more details.

Analog ADC is reported as moved because pin length was fixed to 100mil to comply with KLC but it does not break compatibility. See https://github.com/KiCad/kicad-symbols/pull/1167 for more details

Converter_ACDC.lib:VTX-214-015-103 I am responsible for that one. Vigortronix clarified that the FG pin in that model is actually a NC so we hided it and moved it out of the way. See https://github.com/KiCad/kicad-symbols/pull/1140 for more details.

Power management stuff comes from https://github.com/KiCad/kicad-symbols/pull/572 which I had no time to follow. There is a following PR that we may want to have a look at https://github.com/KiCad/kicad-symbols/pull/1051.

antoniovazquezblanco, 2018-12-05:
Can this be closed?

closed 2018-12-05 by poeschlr