KiCad/kicad-symbols #1217: Questions about the milestones

myfreescalewebpage, 2018-12-04:
Hi,

A few questions relating to the milestones and way Kicad team manage libraries because 3 of my PRs just get in the Backlog milestone.

First, what is the Backlog milestone ? My PRs have been tagged here but with no comment. Do I have to understand nobody take care of them ? Are they accepted ? Are they pending for future review not yet planned ? What's the level of priority of this milestone compared to the PRs with no milestone ?

I looked at the other milestones (https://github.com/KiCad/kicad-symbols/milestone/4 and https://github.com/KiCad/kicad-symbols/milestone/2 today) and I'm really surprised because some PRs are here since several months !
- How do you decide when a PRs should be merged (decision between milestones 5.1.0 or 6.0.0 for example) ?
- I think librarian managers should status on the accepted PRs in a reasonable time, this can create many confusion and another developer can create a PR with a new device that already exists and has been accepted but not yet merged... Also when we submit several modification/additions to the libraries, we create some branches (http://kicad-pcb.org/libraries/contribute/). Developers have several branches that are sometimes difficult to manage when working on a project because the devices are dispatched in these branches. This comment is not for PRs with still some work to do from the submitter and of course I understand that's a lot of job to make the review. I just focuses on the accepted, terminated, but not yet merge PRs and wonder why they are kept open so long time.

At the opposite, I have proposed https://github.com/KiCad/kicad-symbols/pull/1198, accepted and merged only few hours later with no comment. Do I have to understand KLC fixes are encouraged at this time to increase the consistency of the libraries ? That's something I can contribute too but only if your strategy is to go on this way also, because I don't want to spend many time on some libraries without getting merged long time after acceptance.

I was also very surprised by the new Kicad installer, about 1GB to download ! Is it a raisonnable choice ? isn't it better to provide only the software without the libraries and fetch the latest stable release at the first execution ? Separating the software and the libraries seems me a better choice : possibility to choose a specific fork and/or branch of the libraries for the advanced users, possibility to update quickly to the last libraries (git pull or simple button on the Kicad window for non expert users), no need to make a release if you only want to add symbols/footprints with no modification of the software itself, no need to download again 1GB of installer next time to update my installation while most of the content is the same...

Well that's a long comment, hope you have understood it was not a criticism, I really thank you for the job you are making, but only the need for a better understanding you should communicate with people contributing to the project about the roadmap, strategies, etc. Like many others, I do not claim to be a librarian but we would like more visibility to continue contributing the project.

Thanks for reading and comments!

Cheers
Joel

poeschlr, 2018-12-04:
We are only so many volunteers here. If we get a backlog of old PRs then this is because we get more requests than we can handle at that time.
The best way to change this is by helping out with reviewing pull requests. Even without merge privilege you can speed up a lot by checking things for correctness and telling us maintainers that something is ready to be merged. (We will still double check it till we are confident that you can be trusted. At which point you might just be invited to join the team.)

antoniovazquezblanco, 2018-12-04:
@myfreescalewebpage I have been trying to push the usage of tags and milestones into library maintainers for better organization. I am trying new things as we go. I have yet to find a good way to proceed.

My I dea was to try and set currently in review PRs to 5.1.0 release and help push them although some of them have been abandoned. Those I would have to re-make to push getting reviewed. Some of the in review issues cannot be merged before 6.0.0 due to big compatibility changes so those go to that release. Finally, the rest of PRs will get pushed to backlog and would be accepted in the nearest upcoming release if any librarian decides to review it.

Although the library has improved a lot since 4.x releases, I prefer to have a look at fixes of the pending issues but that is just a matter of personal preferences. Other librarians will probably focus on things they are familiar with or just what they feel like learning in a particular moment.

If you really want to contribute to the libraries, as you can see, there is a ton of work to be reviewed and would be awesome if you were able to help us with that.

Thank you for your pointers. If you have any suggestions to improve our workflow they are very welcomed!

evanshultz, 2018-12-04:
Joel,

Regarding your question about the installer, that is something you'd have to take up with the dev team (even though it does relate to the libraries). You can reach them at https://bugs.launchpad.net/kicad.

poeschlr, 2018-12-04:
Regarding installer: There is more than one installer for kicad as it works on many different platforms. When you ask the devs you really need to include the information about the platform in question.

I know for a fact that some linux distros offer separate packages for the libs while others do not. This is because separating packages requires changes to the build scripts. Similarly to the library review this relies on volunteers. (I am sure nobody would be against offering a separate package for the libs under any platform. Especially if you are the volunteer implementing it into the build scripts for your platform of interest.)

poeschlr, 2018-12-04:
There is only one level of official librarian. However every github user can comment on any pull request or issue. Meaning if you see something where you can help out (in any project) you can do that.

We do not really have any other documentation than the KLC. Most rules can be ignored if there is a good reason for it. (Example: a datasheet requests something that would violate KLC. Or if two rules kind of contradict each other. An example for the later would be the requirement to put pin on 100 mil grid and having the symbol centered. The center requirement is the weaker one in this case.) 
If you stick around a bit you will most likely learn how we handle these sort of things. If you are unsure simply ask for help by one of the more experienced people.

The KLC stuff is mostly checked by the script. (We however need to sometimes help new contributors along the way to find out what went wrong here.)
Most of the work for us is checking contributions against the datasheet. For symbols this means checking every pins name, number and type. For footprints it means checking all dimensions against the datasheet and if necessary industry standards (for all parts this means checking the size and position of pads, the fab outline and the courtyard layer. For some parts this also means checking the paste and mask layers.). For footprints one would also need to check the pin 1 marker as this can not be automated. Checking footprints is made a lot easier by kicad-stepup and working with powerful tools in freecad. A tutorial about that can be found here: https://forum.kicad.info/t/how-to-check-footprint-correctness/9279

myfreescalewebpage, 2018-12-04:
> However every github user can comment on any pull request or issue. 

I know.

What about the Labels ? I'm not able to add one on my own PRs...
How to you tag the PR "ok I think this is ready to merge" ?

myfreescalewebpage, 2018-12-04:
@DanSGiesbrecht okay this sound more clear for me. Let me making a scenario below:

1- Submitter make a PR
2- One reviewer add "Pending reviewer" plus "Enhancement" or "Addition" (depending) tags
----- the PR is waiting
3- One reviewer assign itself to the PR and remove the "Pending reviewer" tag
4- The assigned reviewer check the PR (with the datasheet, KLC, etc etc) and comment the PR to get some fixes from the submitter if required
----- the PR is ready to go
=> then ?? which tag to lock the PR/indicate ok for merge to a guy allowed to merged ?

Also I have still not understood Backlog or not Backlog...

> My personal goal is to self-assign to show intent, and review within a week.

Yes thanks. You also added milestone 5.1.0, why ? Why not backlog milestone ? If there is many fixes to be made on the PR, this will not be ready for 5.1.0...

Sorry for all those stupid questions but the process was never very clear since years and I start only now to understood...

myfreescalewebpage, 2018-12-04:
@poeschlr Library changes to come between v5 and v6 ?

poeschlr, 2018-12-04:
Right now no changes to symbols should be made that break old schematics. (moving around pins, removing symbols, ...)
After v5.1.0 is released we might opt to lift this restriction. This would however mean that v5 will not really get further lib updates after that fact.

If we do that then i would still not allow major reorgs. This is such that we have the option to include bugfixes in version 5 with a bit of effort.

myfreescalewebpage, 2018-12-06:
@poeschlr @DanSGiesbrecht so, what about the scenario above ?

poeschlr, 2018-12-06:
I am not really sure what your question is. What scenario to you mean?

myfreescalewebpage, 2018-12-06:
1- Submitter make a PR
2- One reviewer add "Pending reviewer" plus "Enhancement" or "Addition" (depending) tags
----- the PR is waiting
3- One reviewer assign itself to the PR and remove the "Pending reviewer" tag
4- The assigned reviewer check the PR (with the datasheet, KLC, etc etc) and comment the PR to get some fixes from the submitter if required
----- the PR is ready to go
=> then ?? which tag to lock the PR/indicate ok for merge to a guy allowed to merge ?

poeschlr, 2018-12-06:
I think you will find that in that case you are already stuck in step 2. I highly doubt you can assign yourself to a pull request on a repo where you do not have write permission. I am not even sure if you can set tags. (I seem to remember reading somewhere that only members are able to set tags.)

Meaning: The pull request does not have a valid reviewer until one of the members takes it on. (The meaning of "pending review")
If a member takes on a pull request that has a review by a non member then they can either double check that review (highly encouraged) or merge immediately. 
Even if you are not able to participate fully your review will still be helpful to us.. (The reason for this is that checking something is quite fast. Documenting problems and communicating this with the contributor is what takes up a lot of our time.)

Notifying us can be done by simply adding a comment to the pull request. (If a librarian has been assigned then you can even tag them in the message.) So if you think a contribution is ready to merge or if you need further input simply add a comment. (We get a notification for every message left in any pull request. At least on the dashport. I only get mails for things where i am mentioned or where i already contributed)

DanSGiesbrecht, 2018-12-10:
@myfreescalewebpage :
> => then ?? which tag to lock the PR/indicate ok for merge to a guy allowed to merge ?

Any member (librarian) can merge the PR. Typically in your scenario, this would be done by the person assigned (who should be a librarian). If you think that a PR is ready to be merged (librarian's review is complete, any requested changes have been made) and no action has been taken, you can "@" message the member to ask if they can merge the PR.

myfreescalewebpage, 2018-12-10:
@DanSGiesbrecht 

Understood, thanks.
Summary is:
1- Submitter make a PR
2- One reviewer add "Pending reviewer" plus "Enhancement" or "Addition" (depending) tags
----- the PR is waiting
3- One reviewer assign itself to the PR and remove the "Pending reviewer" tag
4- The assigned reviewer check the PR (with the datasheet, KLC, etc etc) and comment the PR to get some fixes from the submitter if required
5- Once the PR is okay, the reviewer merge the PR.

What about the milestones ?

Is it possible to add me as a librarian, so I can process some PRs according to this scenario above (not my PRs of course to get a double checking from another people !) ? Actually I can't change the tags and I can't merge. Is there any people to double check I'm a good librarian before I merge, at least for the first PRs ?

DanSGiesbrecht, 2018-12-10:
Your summary is correct. The milestones are something that @antoniovazquezblanco is rolling out to help track what PRs are aimed to be included before specified releases (e.g. 5.1.0, 6.0.0). My understanding is that currently, `Backlog` refers to any existing PRs that have not been assigned to anyone, and `5.1.0` milestone PRs are currently undertaken by a librarian.

With regards to becoming a librarian, note @poeschlr 's comment above:
>If your reviews are of high quality then you will simply be added to the team.

DanSGiesbrecht, 2018-12-10:
Unless you have further questions about milestones, can we close this issue?

myfreescalewebpage, 2018-12-11:
Okay, understood about the milestones, that's more or less a guideline to treat some priority PRs.

> If your reviews are of high quality then you will simply be added to the team.

Okay, not very convenient to work without moving the tags etc but why not I will start.

closed 2018-12-11 by myfreescalewebpage