KiCad/kicad-symbols #1249: ESP32 Symbol

gcormier, 2018-12-11:
GPIO 34/35 are INPUT in the current symbol, but should be BIDIRECTIONAL as they are GPIO.

https://www.espressif.com/sites/default/files/documentation/esp32-wroom-32d_esp32-wroom-32u_datasheet_en.pdf

[image]

[image]
[image]

DanSGiesbrecht, 2018-12-11:
When I merged this, I defaulted to the datasheet listing those pins as inputs:
[image]

I agree that they appear to be GPIO (bidirectional). If this is a datasheet error, I think that we should change the pins to bidirectional. However, I'm not sure if it's a mistake on Espressif's part, or not.

DanSGiesbrecht, 2018-12-11:
Not a problem! Glad we got it sorted out.