KiCad/kicad-symbols #1254: Unbounded power pins for multi-unit symbol (LM2904)

DanSGiesbrecht, 2018-12-11:
LM2904 in Amplifier_Operational has power pins on Unit C, with no bounding box:
[image]

Example of bounding box from the KLC can be found [here](http://kicad-pcb.org/img/klc/S3.8a.png).

I've pre-booked this for V6, because re-centering/moving the pins breaks compatibility.

evanshultz, 2018-12-11:
This is as desired. See https://github.com/KiCad/kicad-symbols/issues/190 and https://github.com/KiCad/kicad-symbols/issues/451#issuecomment-385770534 and https://github.com/KiCad/kicad-symbols/pull/578. All multi-unit triangle symbols (opamps, comparators, etc.) are like this.

I'm removing the labels and milestone, at least until we decide if any change is needed.

I suppose the pins could be moved to the Y axis, but I just copied them from their old placement.

evanshultz, 2018-12-11:
Yes. I didn't feel that single-gate symbols very totally resolved as Jan proposed. And the logic gates are a similar issue. Probably driven to resolution on all outstanding points would be best to make sure we cover all bases, but I suppose at least putting some of it into KLC would be a good start.

Did you have input on the linked items? The conversation started a while ago so you may have some fresh ideas to incorporate.

evanshultz, 2019-02-22:
Closing this because the conversation that covered this lives elsewhere (see above) and splitting it up loses history. I think that's OK but if you object you can re-open this to add more info.

closed 2019-02-22 by evanshultz