KiCad/kicad-symbols #1273: #1232 post-merge review issues

aewallin, 2018-12-17:
#1232 had some comments made after it was merged:
- The description should end in the package and pin count, TSSOP-16 for this one.
- The description has a leading space.
- Pin length should be 20mil. (original author maybe means 200mils?)

there was also discussion if three separate VDD pins are needed or if they should be stacked. This discussion can continue here.

evanshultz, 2018-12-17:
Stacking of power pins is to be done if the manufacturer requests some unique handling of the pins, such as a decoupling capacitor on each individual pin. If not, the pins should be stacked. This is official stance by Rene, the head librarian.

I can understand common practice may be to add unique decoupling, but unfortunately common practice isn't always common. And library consistency, according to the above rule, was deemed of high importance. In the middle of the v5 release cycle I would expect symbol style guidelines to be maintained.

Is there such documentation for unique decoupling on all VDD pins of this clock buffer? I didn't see it in the datasheet but I may have missed something.

myfreescalewebpage, 2018-12-30:
PR merged, closing the issue.

PR #1316 (2018-12-30): Fix for #1273
This fixes #1273
- improved description
- pin lengths all 100mil
- VDD pins stacked.

[image]
commit 57188b09 (2018-12-30): fix #1273: all pin lenghts 100mil. stack VDD pins.
commit d0561628 (2018-12-30): fix #1273: remove leading space from description. add footprint to end of description
closed 2018-12-30 by myfreescalewebpage