KiCad/kicad-symbols #1289: APDS-9301 is obsolete

antoniovazquezblanco, 2018-12-19:
See:

https://www.mouser.es/ProductDetail/Broadcom-Avago/APDS-9301-020?qs=%2fha2pyFadugHWmphw2NMso7M8viQE%2fwBB4qsckozE8VCYjHIXAsXjg%3d%3d

https://es.rs-online.com/web/p/sensores-de-luz-ambiental/7142722/?relevancy-data=636F3D3126696E3D4931384E53656172636847656E65726963266C753D6573266D6D3D6D61746368616C6C7061727469616C26706D3D5E5B5C707B4C7D5C707B4E647D2D2C2F255C2E5D2B2426706F3D31333326736E3D592673723D2673743D4B4559574F52445F53494E474C455F414C5048415F4E554D455249432673633D592677633D4E4F4E45267573743D415044532D39333031267374613D415044532D3933303126&searchHistory=%7B%22enabled%22%3Atrue%7D

The component should be moved to the obsolete folder

evanshultz, 2018-12-19:
It looks like this part's status is EOL (end of life), and probably NRND (not recommended for new designs), but not yet obsolete.

antoniovazquezblanco, 2018-12-20:
True. My mistake. This got me thinking if support for product lifecycle status should be integrated into the new in-design library format.

closed 2018-12-20 by antoniovazquezblanco