KiCad/kicad-symbols #1317: FT231XS RTS is marked as input.

gantzm, 2018-12-23:
In Interface_USB the symbol FT231XS has pin 2 (RTS) marked as input when it should be marked as output.  I believe this same issue might exist for other symbols in that family.

Is the expectation that I submit a pull request for this or is this something a maintainer would easily fix?

evanshultz, 2018-12-23:
Confirmed by page 10 of the datasheet at https://www.ftdichip.com/Support/Documents/DataSheets/ICs/DS_FT231X.pdf.

Submitting a PR is the best way. We're happy to get fixes for bugs, though we're also sad when they're found.

I see this is your first post at KiCad, so thanks for taking the time to post this issue and volunteer to submit a PR to fix it! In case you hadn't seen it yet, http://kicad-pcb.org/libraries/klc/ describes the guidelines for the KiCad library.

gantzm, 2018-12-23:
I've forked the repo and made a branch.  How do I open the library in the Symbol Library Editor?  When I try to add it it tells me it already exists, which makes sense because it is installed.  Can I edit a stand alone library or do I need to remove the system installed library from a path and add the cloned one?

poeschlr, 2018-12-23:
One option is to add it to the local library table of a project only used for working on libraries.
Another is to add it with a different nickname (click away the warning about the existing library. Then you should be able to edit the nickname of the library.)

Another option is to change the KICAD_SYMBOL_DIR variable to point to your personal fork. (This makes the most sense if you plan on doing stuff in the future. But is also the most dangerous to your own project as you might forget to point the variable back to the system libs once you are done.)

gantzm, 2018-12-23:
A pull request has been submitted.  Thanks for the help.

antoniovazquezblanco, 2019-01-05:
Please reference that PR here if you can to keep track of the progress. Thank you!