KiCad/kicad-symbols #1354: Cypress symbol issues

evanshultz, 2018-12-30:
Seen at https://github.com/KiCad/kicad-symbols/pull/1353.
```
Checking symbol 'CYBL10x6x-56LQxx':
  Violating S3.3
    Symbol outline and fill requirements
    Component outline is thickness 0mil, recommended is 10mil
  Violating S3.6
    Pin name position offset
    Pin offset not preferred value
     - Pin offset (40) should be 20mils unless required by symbol geometry
Changed symbol 'MCU_Cypress.lib:CY8C4xx7LQI-4xx'
Checking symbol 'CY8C4xx7LQI-4xx':
  Violating S3.3
    Symbol outline and fill requirements
    Component outline is thickness 0mil, recommended is 10mil
  Violating S3.6
    Pin name position offset
    Pin offset not preferred value
     - Pin offset (40) should be 20mils unless required by symbol geometry
```

jneiva08, 2018-12-31:
Fixed KLC S3.3 and S3.6, still gives error S4.4 in a few footprints

myfreescalewebpage, 2019-01-02:
@jneiva08 these symbols have other defects, such as VREF pin, datasheet link broken etc. However, your PR fix this issue so I propose to open a new PR if we want to fix other problems.
Joel

closed 2019-01-02 by myfreescalewebpage