KiCad/kicad-symbols #1364: one lib file per symbol? (Digikey library style)

aewallin, 2019-01-03:
When there are multiple pull-requests to the same .lib file, and one PR is merged, we usually get a merge conflict that is not so nice to merge by hand.

The digikey library seems to use a one-lib-file-per-symbol approach, with some scripting (that I didn't find) to generate the final .lib file with many symbols as output.
https://github.com/digikey/digikey-kicad-library
small lib-files seem to be in the /src directory. Not sure how .dcm files are handled.

If the move to another symbol library format is still a year or more away maybe this is a temporary solution that would significantly reduce the need for manually resolving the merge conflicts we have now?

myfreescalewebpage, 2019-01-03:
Had a look to the content of the digikey libs, there is not always 1 symbol per lib.
Also, this is a problem with the alias because multiple parts will need to refer to a single lib.
I don't know what is the kicad roadmap on this subject but too much fiels is also a problem with some operating systems / file systems.

Joel

fauxpark, 2019-01-05:
Each .lib file requires its own entry in the sym-lib-table. The Digikey libraries don't do this because they're not part of the default libraries and the idea is to add them on an as-needed basis per project. Having one symbol to a .lib file would make the sym-lib-table file waaay too big.

Unfortunately, we just have to wait for the new file format to be introduced, which I believe will look something like the current footprint file format and directory structure.

aewallin, 2019-01-05:
Does anyone know the roadmap - is there a new fileformat in the pipeline and how far out is it, 1 year, more?

antoniovazquezblanco, 2019-01-05:
I do not have information about the timeline but here you can find the development mailing list thread about the topic: https://lists.launchpad.net/kicad-developers/msg38760.html

myfreescalewebpage, 2019-03-15:
@aewallin do you think we can close the issue regarding to the title (original question was about the library style) ?

aewallin, 2019-03-16:
I agree, this can be closed.

closed 2019-03-16 by aewallin