KiCad/kicad-symbols #1365: MCU_Microchip_SAMD local uncommitted changes??

aewallin, 2019-01-03:
Recently I'm getting a strange git error with MCU_Microchip_SAMD.lib and MCU_Microchip_SAMD.dcm

after a pull from upstream master I get local uncommitted changes in these files although I have never touched those parts or libraries??

poeschlr, 2019-01-03:
Run diff. does it say something about crlf problems?

myfreescalewebpage, 2019-01-03:
This new lib I have recently merged. KiCad do not complain on my side.

Joel

myfreescalewebpage, 2019-01-03:
> the merge with possibly CRLF issues is #1107 ?

Yes.

I'm working on Windows, probably the submitter of this PR too. And you are probably on Linux ? The creation of some files may be a problem with CRLF yes.

aewallin, 2019-01-04:
linux machine at work had no issues with this?

on another linux machine I am still stuck with the CRLF issue :(

myfreescalewebpage, 2019-01-04:
I know git has configuration options about the line ending but I'm not an expert, a starting point to compare your two machines may be : https://help.github.com/articles/dealing-with-line-endings/

aewallin, 2019-01-04:
meh, on my problematic machine I upgraded from git 2.7.x that comes with Ubuntu LTS16.04  to the latest available from https://launchpad.net/~git-core/+archive/ubuntu/ppa  which is 2.20.1

now the CRLF problem is gone.

closed 2019-01-04 by aewallin

myfreescalewebpage, 2019-01-04:
Great !