KiCad/kicad-symbols #1369: LM358Z-2.5 in Reference_Voltage

davidsrsb, 2019-01-04:
LM358 is a dual opamp.
I think that this is a typo for LM385Z-2.5

myfreescalewebpage, 2019-01-04:
@davidsrsb probably I have missed one episode, but what do you mean ? Wrong name ? wrong device ? wrong lib ?

Cheers
Joel

davidsrsb, 2019-01-05:
PR created
https://github.com/KiCad/kicad-symbols/pull/1376

davidsrsb, 2019-01-05:
> Understood, device "LM358Z-2.5" in Reference Voltage library should be
> renamed "LM385Z-2.5", I agree. Do you want to submit the PR ?

I am confused by the Travis error referring to there being no metadata
properties. This symbol is an alias and has no properties of its own

myfreescalewebpage, 2019-01-17:
PR re-pushed at https://github.com/KiCad/kicad-symbols/pull/1383.

Merged, closing this issue.

closed 2019-01-17 by myfreescalewebpage