KiCad/kicad-symbols #1377: 74LVC1GU04 seems wrong.

matthuszagh, 2019-01-05:
I want to use [SN74LVC1GU04](http://www.ti.com/lit/ds/symlink/sn74lvc1gu04.pdf) in a design and noticed 74LVC1GU04 in the libraries that seems to be the same component. A screenshot of the library component is shown below.

[image]

Assuming this is the same component, I'm confused about a few things.
1. Why aren't the gnd and vcc pins made visible? As it is, I can't connect to them.
2. The footprint filter doesn't suggest 5-pin footprints, such as SOT-553. The datasheet specifies several 5-pin footprints and only a single 4-pin footprint.
3. The datasheet specified in the library is a broken link so I can't check that to make sure this is the same component.

Thanks.

aewallin, 2019-01-05:
This is discussed in #756 

Hidden power pins are not recommended for new symbols, and current symbols in the library should be fixed eventually.
I guess the idea for how to use them is to create named global nets e.g. VCC and GND elsewhere on your schematic, and then the power pins are just magically connected to the same net.

for your 2. question I think the trend is towards an 'atomic' library, so each symbol should correspond to a single footprint. The generic 74LVxx symbol would be split into several 74LVxxYY symbols where YY is the part-number specifying the package, e.g SOT-553.

So the way forward is that you take the existing symbol, make the power pins visible, figure out what part-number the 4-pin package you want is, and submit that as a new symbol PR.

matthuszagh, 2019-01-06:
@aewallin, to clarify, should I remove the old symbol, or just add the new one?

aewallin, 2019-01-06:
sorry I don't know - @poeschlr : can hidden-power-pin symbols be removed/edited now, breaking current schematics, in preparation for a 5.1 release?

matthuszagh, 2019-01-06:
I've made [a PR](https://github.com/KiCad/kicad-symbols/pull/1386) for this without removing the old component. Let me know if you'd like me to do that and I will.