KiCad/kicad-symbols #1400: Where to place new inverter symbol

matthuszagh, 2019-01-09:
I'm planning to add [NC7SVU04](https://datasheet.octopart.com/NC7SVU04P5X-ON-Semiconductor-datasheet-143664.pdf) to the symbol library but am unsure where to place it. Does it make sense to add a Logic_Inverter or Logic_Gate library? Is there somewhere this sort of thing is already placed and I missed it? Thanks.

myfreescalewebpage, 2019-01-09:
Would like to vote for "Logic_Gate".

evanshultz, 2019-01-09:
It fits best with the other single-gate logic symbols at `74xGxx.lib`. Adding a new library requires updating sym-lib-table, and that is not updated during a KiCad installation so nobody would see this new library unless they manually update their symbol table file. Plus, we generally build up a corpus of parts before splitting off and starting a new library.

During the v5 development cycle I vote for consistency and we already have a suitable library. Even if this isn't strictly a 74xGxx part, it still fits there and the much-improved symbol searching means it will be easy to find regardless of where it goes.

The logic libraries all need a massive update which I hope will be included in v6. A re-name that moves all logic symbols to `Logic_xxx` was the direction of some previous discussion so at the right time that's what I will support.

matthuszagh, 2019-01-09:
Makes sense, thank you Joel/Evan. I'll put it in the existing lib.

myfreescalewebpage, 2019-01-09:
Closing the issue since the PR has been made with symbol at the correct location.

closed 2019-01-09 by myfreescalewebpage