KiCad/kicad-symbols #1418: Issues with LP5907MFX footprint.

matthuszagh, 2019-01-11:
The current footprint for LP5907MFX-1.2 and aliases is Package_TO_SOT_SMD:SOT-23-5. However, based on [the datasheet](http://www.ti.com/lit/ds/symlink/lp5907.pdf) (page 37) I believe this is incorrect. From the datasheet:

[image]

Pads in the currently existing footprint file are separated horizontally by 2.2mm whereas the datasheet specifies 2.6mm. Additionally, the pad xy dimensions of 1.06x0.65mm differ from the datasheet's 1.1x0.6mm. The 1.8mm body width is larger than the datasheet's 1.6mm and body length of 3.1mm is larger than the 2.9mm. The vertical pad separation seems fine. I'm going to open a PR with this change, but any comments/concerns are appreciated.

evanshultz, 2019-01-11:
This may be best off in the footprint repo, but because it could affect some symbols I understand why you may have put it here.

More important than the footprint is the package. Our scripts built footprints for many parts based on the package dimensions and IPC rules.

Notice on page 40 of the datasheet that note 5 intimates that IPC 7351 may have an alternate design. This indicates that an IPC footprint will be acceptable but no objection is given. Also see that this is an older design because 7351B has been out for a number of years and 7351C has some significant work done already.

Note 3 on the previous page tells us this is a standard package design which matches a JEDEC drawing (I checked and it does indeed match).

OK, moving back to our current footprint, there is no attribution for https://github.com/KiCad/kicad-footprints/blob/master/Package_TO_SOT_SMD.pretty/SOT-23-5.kicad_mod so we can't really know how or why it was created (I didn't take the time to dig through Git history to see if more info is there).

So... I don't see any reason why this needs a custom footprint. Using the existing scripts to update the existing SOT-23-5 footprint, based on JEDEC MO-178 dimensions, with IPC 7351B parameters with rounded rect pads (from 7351C), is the right way to go. That should work for this and all SOT-23-5 that aren't some goofy design that doesn't stick with the industry standard. And the existing part gives nothing in the footprint name that would make a user think it is non-standard.

The resolution is then adding SOT-23-5 dimensional info to the script input files and then verifying the resulting footprint. This should work for many more than just the SOT-23-5 footprint as well.

Would you please review the above and add any comments. Unless you have some surprising information, the above is the direction we should take and you may close your existing PRs on this matter and you're welcome to jump into the scripting if you like. One of the librarians will get to this in the not-too-distant future, I hope.

matthuszagh, 2019-01-11:
This is very helpful, thanks for all that information @evanshultz. I've closed the related PR's and I'll take a look at the scripting for this. Is there a way to get the IPC information? It looks like you need to buy those documents.

evanshultz, 2019-02-22:
Closing based on the conversation above. Scripting SOD/SOT footprints is likely coming relatively soon since other issues have been noted, but after 5.1 is released.

closed 2019-02-22 by evanshultz