KiCad/kicad-symbols #1422: TPS2054D: Pin 14 ("OUT2") is marked as power input instead of power output

sre, 2019-01-11:
Hi,

TPS2054D's pin 14 ("OUT2") has type "power input" instead of "power output". Power Output makes more sense and is already used by OUT1, OUT3 and OUT4. I suppose this was a typo.

https://github.com/KiCad/kicad-symbols/blob/bfd557726263fea70079776946d08cd35a210480/Power_Management.lib#L1765

Thanks for taking care of the library,

-- Sebastian

evanshultz, 2019-01-12:
http://www.ti.com/lit/ds/symlink/tps2044.pdf

I agree that Power Input doesn't make sense for pin 14.

I would argue all IN and OUT pins should be Passive since we (owners of the symbol) have no idea what signal is going through the part. This is the way our more recent analog switch symbols are made for just this reason.

While this switch is also noted as being bidirectional, there could be a body diode which would make the switch best used in only one directional. It is suggested for use with power, but that still doesn't mean it MUST be used to supply power and thus the OUT pins should be Power Output type.

Thoughts?

closed 2019-01-14 by evanshultz

myfreescalewebpage, 2019-01-15:
Thanks @sre, good catch and good job!

PR #1427 (2019-01-14): TPS2054D: Make IN/OUT pins passive
OUT2 was accidently marked as power input instead of power output, which
resulted in ERC errors with the chips default use case (multi-port USB
current limit). Since the chip could be used for different applications
and is even bidirectional, this patch changes all IN/OUT pins to use the
passive type.

Fixes: #1422
commit 90063063 (2019-01-14): TPS2054D: Make IN/OUT pins passive (#1427)

OUT2 was accidently marked as power input instead of power output, which
resulted in ERC errors with the chips default use case (multi-port USB
current limit). Since the chip could be used for different applications
and is even bidirectional, this patch changes all IN/OUT pins to use the
passive type.

Fixes: #1422