KiCad/kicad-symbols #1440: incorrect pinout for Diode_Bridge:W005G

ppelleti, 2019-01-19:
Here is the footprint for `Diode_THT:Diode_Bridge_Round_D9.8mm`:

[image]

Here is a diagram from the [datasheet](http://www.vishay.com/docs/88769/woo5g.pdf) referenced by the footprint:

[image]

By looking at these two images, I would conclude that the pinout is:

```
1 ~
2 +
3 ~
4 -
```

However, here is the schematic symbol `Diode_Bridge:W005G`, which references this footprint and datasheet:

[image]

According to this symbol, the pinout is:

```
1 +
2 ~
3 -
4 ~
```

So, it appears to me that something is wrong.

jkriege2, 2019-01-19:
I think it is correct ... look at this view from the datasheet you linked:
[image]

Also have a look at the photo here: https://uk.rs-online.com/web/p/bridge-rectifiers/6295153/

The lower drawing in your screenshot is the bottom-view I think ... so then the pinout should be OK? Can you countercheck?

ppelleti, 2019-01-19:
> The lower drawing in your screenshot is the bottom-view I think ... so then the pinout should be OK? Can you countercheck?

Yes, you're right.  Sorry about that.

closed 2019-01-19 by ppelleti