KiCad/kicad-symbols #1450: New library for haptic driver ICs

mpkopec, 2019-01-22:
Hello,
I would like to add DRV2510 and DRV2511 haptic coil drivers to the library, but the closest is driver_motor. The problem is, that although being the best fit, it is still not very good. Do you think starting a new library is a good idea? I know that a lot of LRA drivers and piezo haptic drivers could also go in there. Also, active haptics is a trending topic, so IMO the component number of this type will rise.

Have a nice day,
Maciej

myfreescalewebpage, 2019-02-08:
@poeschlr @evanshultz maybe your opinion on this subject : is "Driver_Haptic" a good library name for this device ?

Cheers,
Joel

myfreescalewebpage, 2019-02-12:
@mpkopec seems other librarians are not perturbed by this new library, I think we can accept it according to me.

closed 2019-02-12 by myfreescalewebpage