KiCad/kicad-symbols #1514: Datasheet links from WCH

evanshultz, 2019-02-11:
CH330/340 and CH376 have less than stellar datasheets. This is an issue to check on this at the next major release and see if better English datasheets are available.

myfreescalewebpage, 2019-02-12:
@evanshultz what do you think of the followings:
- https://cdn.hackaday.io/files/1623896947650976/CH330DS1.zh-CN.en.pdf
- https://www.mpja.com/download/35227cpdata.pdf
- https://www.mpja.com/download/ch376ds1.pdf

Joel

evanshultz, 2019-02-12:
I'm hoping WCH has good, up-to-date datasheets, so I picked the next major release to revisit this topic. Any third party datasheet is less than official by definition.

myfreescalewebpage, 2019-02-13:
Sure. Above are in English and valid, which is not currently the case for the three parts, not found any on WCH website itself.

evanshultz, 2019-02-13:
See the symbol merge PRs for some additional discussion and why we ended up using the datasheets we did, at least for the symbols I've been involved with.

myfreescalewebpage, 2019-02-13:
Read, understood! PR for future reference: https://github.com/KiCad/kicad-symbols/pull/1047 and https://github.com/KiCad/kicad-symbols/pull/814

evanshultz, 2019-02-13:
I was lazy and should have looked them up myself. I have also been in contact with WCH and have been told (we will see!) that improved/updated English datasheets are going to happen.