KiCad/kicad-symbols #1519: New Geiger-Müller symbols

heitorPB, 2019-02-13:
We had a discussion in the forum about Geiger-Müller symbol [here](https://forum.kicad.info/t/geiger-tube-symbol/15058/). I came up with a symbol based on Google image searches. #1492 added it and #1502 moved it to the right place.

User 9V1MI in the forums found IEEE 315 that specifies how the symbol should be and paulvdh drawed these two:

[image]

We have some questions on how to proceed now:

- [ ] The correct name is `Geiger-Müller`, but the letter `ü` is not allowed by the KLC. There was a comment by @evanshultz [here](https://github.com/KiCad/kicad-symbols/issues/1470#issuecomment-462957116) to use the umlaut `ue` instead: `Geiger-Mueller`. I guess it would not be a problem for people that use `u` instead of `ü`?
- [ ] How should be the names of the other 2 alternative symbols? `Geiger-Mueller_alternative` and `Geiger-Mueller_alternative2`?
- [ ] How should be the names of the pins? `1` and `2`? `E1` and `E2`? `A` and `C` (anode/cathode)?
- [ ] Just to be sure: they belong in the library `sensor`, right?

poeschlr, 2019-02-13:
What is the difference in function for the generic and normal variant? (Device function. I see a dot between pin 2 and housing what does this mean?)

Why do we need alternative versions at all? (How do they look like. What difference in device function to they communicate?)

Numbered pin names are generally preferred.

Sensor is a good library for these.

heitorPB, 2019-02-13:
> What is the difference in function for the generic and normal variant? (Device function. I see a dot between pin 2 and housing what does this mean?)

The dot means that the enclosure is metallic and connected to terminal 2.

The current symbol in the lib is this:
[image]

I made it based on schematics I found in internet. And it doesn't say anything about the enclosure.

evanshultz, 2019-02-13:
Thanks for moving this to an issue.

1. Since IEEE 315 shows the symbols in the first post I would prefer just using that one. (Also, the symbol above is missing a dot just above pin 2.) While we're a bit loose with symbol styles we have generally stuck with IEEE 315 if it gave good guidance that we knew was supported by our experience.
1. IEEE 315 does show this symbol for other devices under the heading `Nuclear-Radiation Detector`. If this symbol can also apply to Proportional Counter Tube and Ionization Chambers, why should we be so specific and call it `Geiger-Mueller`? While I'm not an expert in this area, from the 5 minutes of reading I've done it seems there's a better and more encompassing symbol name that should be used.
1. Is the dot enough to warrant a separate symbol? For such a rarely-used symbol (compared to all symbols in the library), just the undotted one should suffice IMO.
1. Yes, the umlaut should be replaced. We used Wuerth for this reason already. It is understandable and common for users of languages which contain umlauts.
1. Pin numbers and library are covered above by @poeschlr .

Please respond and, when you're ready, submit a PR to get the changes merged.

heitorPB, 2019-02-14:
> 1. Since IEEE 315 shows the symbols in the first post I would prefer just using that one. (Also, the symbol above is missing a dot just above pin 2.) While we're a bit loose with symbol styles we [...]

I'm not sure what that dot means. Do you know it?

>     2\. IEEE 315 does show this symbol for other devices under the heading `Nuclear-Radiation Detector`. If this symbol can also apply to Proportional Counter Tube and Ionization Chambers, why should we be so specific and call it `Geiger-Mueller`? While I'm not an expert in this area, from the 5 minutes of reading I've done it seems there's a better and more encompassing symbol name that should be used.

I'm not an expert either. And also I don't have access to this specification. But renaming it to  `Nuclear-Radiation Detector` sounds good. What is the difference between `keywords` and `aliases` in the symbol properties?

>     3\. Is the dot enough to warrant a separate symbol? For such a rarely-used symbol (compared to all symbols in the library), just the undotted one should suffice IMO.

I'm OK with that.

evanshultz, 2019-02-14:
I believe it was already mentioned that the dot indicates one collector is connected to the enclosure.

You can search for IEEE 315-1975 online if you want to see what it contains. Check section 7.7.

Keywords are, well, key words to help users find the symbol they want. The ALIAS feature of KiCad allows one symbol to be shared with multiple parts. For example, a dual opamp symbol could be used for TL072, MC33078, LM358, etc. Each ALIAS has a unique description/keywords/datasheet.

heitorPB, 2019-02-14:
About the dot: one indicates that the enclosure is connected to one terminal. But there's another dot, near the 'plate' of pin 2.

> Keywords are, well, key words to help users find the symbol they want. The ALIAS feature of KiCad allows one symbol to be shared with multiple parts. For example, a dual opamp symbol could be used for TL072, MC33078, LM358, etc. Each ALIAS has a unique description/keywords/datasheet.

So the name should be `Nuclear-Radiation Detector` and some keywords include: `Geiger-Mueller`, `Neutron Counter`, `Proportional Counter Tube` and `Ionization Chamber`. Keywords can include `ü`?

PR #1550 (2019-02-22): Replace sensor/Geiger-Muller with sensor/Nuclear-Radiation_Detector
fix #1519

Symbol on a 50 mil grid:
[image]
commit 5ce1c2e8 (2019-02-22): Replace sensor/Geiger-Muller with sensor/Nuclear-Radiation_Detector (#1550)

fix #1519
closed 2019-02-22 by evanshultz