KiCad/kicad-symbols #1521: pinout for symbol USB6B1 is wrong

mazhe39, 2019-02-13:
The datasheet of USB6B1 has a completely different pinout than the one in kicad (v5.0.2). It appears the symbol for USB6B1 in kicad is copied from USBLC6-2SC6 which has the same functionality but different pinout. 

Chris

antoniovazquezblanco, 2019-02-13:
Would you be willing to create a PR with de corresponding fix?

Thanks!

mazhe39, 2019-02-13:
I'd love to help but I am not sure where to get started. Is there a quick-guide for novices to github?

Chris

On 2019/2/13 下午 06:59, Antonio Vazquez wrote:

Would you be willing to create a PR with de corresponding fix?

Thanks!

evanshultz, 2019-02-13:
PR has already been submitted at https://github.com/KiCad/kicad-symbols/pull/1060. Hopefully we will get a response soon so this can be merged. If not, since this is clearly a bug, I would entertain merging another PR to get it into the 5.1 library.

evanshultz, 2019-02-27:
#1060 has been merged.

closed 2019-02-27 by evanshultz