KiCad/kicad-symbols #1523: Ground connection of 3 pins crystal symbols are not align with 50mil grid

hildogjr, 2019-02-13:
Ground connection of 3 pins crystal symbols are not align with 50mil grid, instead these ground pins are aligned with a 25mil grid.

evanshultz, 2019-02-13:
You are referring to the crystal symbols that end in `_Small`, like this one:
[image]

There are four of them in the library.

http://kicad-pcb.org/libraries/klc/S4.1/ doesn't allow any pins off a 100mil grid, and almost all of the other `_Small` symbols still use a 100mil grid. Only the coupled inductors (like `L_Core_Iron_Coupled_Small`) have pins on a 50mil grid except for these crystals.

KLC allows shorter pin lengths for small discrete components, which apply to the `_Small` symbols, but there is no allowance for pin grid placement. So these symbols are in violation.

Because this makes it hard to connect wires to the pins if they're off grid, I think KLC is correct and these symbols are a problem. I propose changing the coupled inductor and these grounded crystal symbols to put all pins on a 100mil grid. Possible fixes:
1. The coupled inductor symbols can just be shifted up by 50mil.
1. Crystal ground pins can shrink to 25mil in length and be placed at a Y location of +/-100mil to be on grid. The pin number then smacks into the symbol but if the pin number size is changed to 30mil it just clears the symbol geometry. Leaving the pin number text size alone isn't that bad, only a bit ugly.

@poeschlr 
Sound OK to you? Should this be done right away and treated as a bug, or wait until v6?

@hildogjr 
Thanks for catching this issue and reporting it!

hildogjr, 2019-02-13:
I think as a small fix (not change any folder/name) should be interesting already be avaible at v5.1.
Yes, the ground pin could be shrink 25mil or yes better placing it down 25mil and use a line segment?

evanshultz, 2019-02-13:
@hildogjr 
Indeed, I would like to see this fixed in v5.1 as well.

Moving the crystal ground pin down by 25mil puts it at [0, -150], which is not on a 100mil grid. So that's not a fix to me. We want all symbol pins to be on a 100mil grid for wiring.

hildogjr, 2019-02-14:
Moving it down at [0, -200] mil, i think, will make the symbol bigger. Shrink it appear to be the best option, I just concerned about the pin name text.

closed 2019-03-11 by myfreescalewebpage