KiCad/kicad-symbols #1526: Should we be a bit more lenient with the grid for pins?

poeschlr, 2019-02-13:
Right now the KLC and especially the travis check make it seem as if pins need to be on 100mil grid at all cost.
It might however be a good idea to allow at least some exceptions. One example would be not connected pins that we require to be on the symbol boarder. I would say for these the 50mil grid should really be good enough.
I am sure we can come up with other reasonable examples where the 50mil grid would be ok.

I am not prepared to allow anything below 50mil grid for pins.

evanshultz, 2019-02-13:
If the pins of the component are all 100mil long, then NC pins on the symbol outline will have no problem to be on a 100mil grid. The issue that spawned this only has an off-grid failure because the pins should be 100mil long but they were 150mil long. Correcting the pin length resolved the off-grid issue.

I'm not objecting to this clause in KLC as when we do have 150mil long pins, like for packages with >99 pins, this is a reasonable thing to do.

poeschlr, 2019-02-13:
or if some pin numbers include a letter like for bgas ;) Then you get it with >9pins. (And i suspect that bga parts are quite likely to have a few not connected balls.)

fauxpark, 2019-02-14:
For the 100-pin ATmegas I avoided these off-grid errors by placing NC pins on the next 100mil grid point inside the box:

[image]

There is no "requirement" for NC pins to lie on the border - note the word used in S4.6 is "should". Perhaps it could be changed to "The end of the pin should lie on or inside the symbol's outline box (on a 100mil grid)".

evanshultz, 2019-02-22:
Agree. I don't see why we can't keep a strict 100mil grid requirement.

(I wasn't reading your post properly and went off topic above. Sorry. This was pin grid not length.)

calebreister, 2019-04-13:
I've got another potential exception. Both screenshots below are of LVDS driver ICs. The first one is in the official symbol library, and the second I created for a current project. Now, which is easier to read? I would argue that the second symbol more clearly shows the logical flow and design intent. Isn't that the reason we create schematics (instead of just typing in netlists) in the first place?

[image]

[image]

poeschlr, 2019-04-15:
Maybe another reasoning from my side: We only have this rule because of a missing feature in kicad. Meaning as soon as kicad can snap to pins we do no longer need this rule at all. At that point we might switch to some lower grid spacing to still ensure consistent look and feel but be even more lenient about it than now.

I also wonder if it was a good idea to choose the maximum available grid as it really restricts what can be done. The second largest might have been the better choice looking back but i do not want to change this for the whole lib during a stable release. But then again i still think there can be exceptions to that rule.

yankee14, 2020-04-03:
@poeschlr @evanshultz I noticed that many of the symbols in the Device category are not on a 100 mil grid, for example Device-->D_Schottky or even Device-->R. Should I fix these and submit a pull request?

evanshultz, 2020-04-03:
@yankee14 
Nope. For one, changing those would hose up loads of schematics so definitely not now.

I believe https://kicad-pcb.org/libraries/klc/S4.1/ is supposed to capture this, but it's not. The reduction in pin length is clear but allowing 'simple' devices more leeway in pin grid isn't there. Perhaps the best way to capture this would be to state that 'simple' devices need pin spacing on a 100mil grid. Those parts you mention have pins at -150,0 and 150,0, for example, so the pins are 300mil apart and thus it works with a schematic having a 100mil grid. Make sense?

yankee14, 2020-04-03:
@evanschultz Yes it does, thank you. I guess I didn't realize whether the klc enforces all pins lying on the absolute 100 mil grid vs all pins relatively lying multiples of 100 mils from one another.