KiCad/kicad-symbols #1533: Issues with passive filter symbols

evanshultz, 2019-02-14:
This applies to the Mini-Circuits symbols in the RF_Filter library, but there may be other symbols to which this applies.

There should be 100mil long pins on all of these symbols, and that may change size of symbol body to keep the pins on a 100mil grid. Regardless of filter type the symbol design can be consistent. There's no reason that they can't use the recommended pin length from KLC. That goes for NC pins too.

The pins should be Passive type as this component contains only passive parts. It may not matter in most schematics, but there could be a valid circuit that would trigger a false ERC error because of the pin types being Input and Output.

There are overlapping elements on some symbols, notably the word "OUT" and the filter symbol.

Because of the high number of similar symbols, scripting would be a great solution.

aewallin, 2019-08-30:
pin length: I suggest S4.1 can be made clearer - if the desire is to stick to 100mil unless really forced to do otherwise (see especially S4.1.2 "number of characters...")

evanshultz, 2019-08-30:
https://github.com/KiCad/kicad-website/pull/417 made some updates to this clause. Are you saying it still lacks clarity that 100mil is the desired pin length unless a change in length must be made?