KiCad/kicad-symbols #1547: Broken datasheet links for Linear Technology

calebreister, 2019-02-18:
It appears that the datasheet links for most (if not all) of the Linear Technology ICs are broken since they were acquired by Analog Devices in 2017. This may be scriptable, but it may be easier (and more reliable) just to go through the library and fix all of the links manually.

evanshultz, 2019-02-19:
Here are two examples. Search and replace may work.

LTC6362
Was: http://cds.linear.com/docs/en/datasheet/6362fa.pdf
Is: https://www.analog.com/media/en/technical-documentation/data-sheets/6362fa.pdf

LT1054:
Was: http://cds.linear.com/docs/en/datasheet/1054lfh.pdf
Is: https://www.analog.com/media/en/technical-documentation/data-sheets/1054lfh.pdf

evanshultz, 2019-02-23:
Thanks for the report. This has been resolved with https://github.com/KiCad/kicad-symbols/pull/1562 and https://github.com/KiCad/kicad-symbols/pull/1563. Thank you @lorem-ipsum !

closed 2019-02-23 by evanshultz