KiCad/kicad-symbols #1570: Silly question about 74xx hidden power pins

diegoherranz, 2019-02-23:
I know that we want to fix this for v6.

Just a silly question. Now that these symbols use hidden power pins, which become global nets, is there any way of using different power rails for different ICs with the symbols currently on the library?

I think it's not possible and I will need to modify the symbols, but asking just in case I'm missing something.

Thanks!

evanshultz, 2019-02-23:
It's not a silly question. :)

I don't think so, and not being able to select the net connection of the power pins is a problem for many reasons and in many designs. Maybe I'm wrong but that's what I saw (though I haven't used many of these gates outside of testing for issues and PRs).

Probably we should have been more aggressive about pursuing this for v5 or v5.1, but at this point it won't be ready.

diegoherranz, 2019-02-23:
Thanks for the quick reply. I was wondering whether I had missed something.
Agree, too late for v5.1, so let's get it right for v6.
Thanks!

closed 2019-02-23 by diegoherranz

poeschlr, 2019-02-25:
Yes this issue is exactly the reason why we want to change the symbols. It has been a known issue for a very long time. Many people requested this to be changed but nobody was ever prepared to put in the time necessary to fix the library. This is why it was left in the state it is.

diegoherranz, 2019-02-25:
Given how close we are to v5.1, it'll have to be addressed for v6.
Thanks!