KiCad/kicad-symbols #1629: Raspberry Pi - Computer Module

jneiva08, 2019-03-12:
Have idea existed Symbol PR for Raspberry Pi - Computer Module, but can't find.

If needs to be created it should be created on MCU_Module?
It can be only one symbol or it should be separated by output type?

evanshultz, 2019-03-12:
What is the purpose? There is an existing template, but it depends on what you want to do.

The issue of template vs footprint, and what they contain, is currently in flux. See https://github.com/KiCad/kicad-footprints/issues/1307 and other related discussions.

myfreescalewebpage, 2019-03-12:
Maybe @jneiva08 wish to add the module as done for https://github.com/KiCad/kicad-symbols/pull/1195 ?

jneiva08, 2019-03-13:
I'm speaking of this:

[image]

Exist a [PR](https://github.com/KiCad/kicad-footprints/pull/1187) for the footprint from herostrat

I will need to use compute module for my job and for my friend.

evanshultz, 2019-03-13:
I suppose MCU_Module would work. I don't see a more suitable place.

Here is how I'm thinking this would go:
- You could use a regular 200-pin connector symbol and a SODIMM footprint.
- However, having a specific symbol makes the pin functions more clear.
- This symbol could pick a SODIMM footprint as the footprint filter.
- A footprint for the module outline could be added and to the board along with the SODIMM header.
- Or a new footprint could be added that includes the SODIMM pads and the RPi board outline.

I don't know which approach is best. It seems this is not an area the official library has figured out yet (?).

@jneiva08 
How were you planning to approach this?

myfreescalewebpage, 2019-03-13:
@evanshultz  that's exactly what we done together with the i.MX6 module I have proposed in #1195.

@jneiva I think you can:
- submit a specific symbol to detail te behavior of the pins. Something starting from the ViionSOM-6ULL of #1195 is great
- do not specify footprint, but only `*SODIMM*` footprint filter to allow SODIMM200 from several manufacturers to be selected

Joel

myfreescalewebpage, 2019-03-15:
@jneiva08 does it answer your question ? Thanks.

jneiva08, 2019-03-15:
Yes, I hope that during the weekend you can make the most of the symbol

myfreescalewebpage, 2019-03-15:
So I'm closing the issue. Thanks.

closed 2019-03-15 by myfreescalewebpage