KiCad/kicad-symbols #1635: Batteries have passive pins

feklee, 2019-03-13:
I expect pins to be of type *power output*.

Affected symbols: *Battery_Cell* and *Battery* in `Device.lib`

See also Sprig's forum post: [ErrType(3): Pin connected to some others pins but no pin to drive it](https://forum.kicad.info/t/errtype-3-pin-connected-to-some-others-pins-but-no-pin-to-drive-it/10946?u=feklee)

evanshultz, 2019-03-13:
@poeschlr 
How do you want to handle this? It is not always the case that changing this will resolve ERC errors, for example if the battery is being charged or probably some other circumstances. This is exactly the case for having the PWR_FLAG. But maybe there were decisions I don't know about or recall.

feklee, 2019-03-13:
Battery being charged I did not think about. My bad. (As a lame excuse, in German we differentiate between [Batterie](https://de.wikipedia.org/wiki/Batterie_(Elektrotechnik)) and [Akku](https://de.wikipedia.org/wiki/Akkumulator), only the latter being rechargeable.)

myfreescalewebpage, 2019-03-14:
@feklee the symbol is the same in KiCad so staying with passive handle every possible combinaisons.

feklee, 2019-03-14:
It's fine for me, although a little bit ugly with the need to use `PWR_FLAG`. In the aforementioned [forum thread](https://forum.kicad.info/t/errtype-3-pin-connected-to-some-others-pins-but-no-pin-to-drive-it/10946) various solutions were discussed. In fact in my current project I have the (rare) case of two batteries in series and on different boards. So having both pins as *power output* would lead to another ERC issue. Didn't think about that before.

evanshultz, 2019-03-14:
Yes! This is exactly why the PWR_FLAG exists!!

Rene, the head librarian may want to make a change but I don't see a problem. It may not be everybody's preference, but this is the construct that the KiCad devs have set up so the library follows.

myfreescalewebpage, 2019-03-14:
@feklee @evanshultz if this is satisfying, I think we can close the issue :)

closed 2019-03-15 by myfreescalewebpage

feklee, 2019-03-20:
> Yes! This is exactly why the PWR_FLAG exists!!

Regarding the `PWR_FLAG` – as I've already done in the the [forum](https://forum.kicad.info/t/erc-error-with-simple-battery-led-circuit/15718?u=feklee) – I want to point out: I find it very confusing that a power symbols such as `+5V` is described as *power flag* as well. Semantically it doesn’t make sense to me that I have to add a power flag to a net that already has a power flag.

[image]

antoniovazquezblanco, 2019-03-20:
@feklee Could you please open a PR fixing the description of those syms?

Thanks!

evanshultz, 2019-03-20:
@antoniovazquezblanco 
Let's provide some guidance. What would you suggest changing the description to?