KiCad/kicad-symbols #1692: Symbol for Connector with more then 40 Pins?

daviid5, 2019-03-28:
Are there any Symbols for Connectors with more then 40 Pins?
I want to use the "Hirose_FH12-50S-0.5SH_1x50-1MP_P0.50mm_Horizontal", but is there any symbol, which I can user for this 50 Pin connector?
I can't find any :(

evanshultz, 2019-03-28:
We have a bunch of them:
[image]

poeschlr, 2019-03-29:
@evanshultz these are for dual row connectors not for single row ones like the FTSH linked to by @daviid5.

So yes technically one could use that dual row symbol but i feel it is confusing to readers of the schematic. Additionally the footprint filter of the dual row symbol is specifically designed for dual row footprints.

Meaning i agree that we might need to add some more symbols to be able to use all FFC footprints we have in the lib. We might however need to split it into multiple units at some pin number cutoff point.

evanshultz, 2019-03-29:
OK. Closing.

BTW, a quick look at DigiKey shows there are FFC/FPC connectors up to [136 pins](https://www.molex.com/pdm_docs/sd/5025551360_sd.pdf) (not that we need symbols for all of them, of course).

closed 2019-03-29 by evanshultz