KiCad/kicad-symbols #1704: Tristate vs Bidirectional MCU IO pins

rs-development, 2019-03-31:
Some MCUs have bidirectional IO pins, some have tri-state.

I am facing ERC errors when connecting power outputs, e.g. reference voltages to analog reference pins. which are of type bidirectional.

What is the correct type? What should i use in this kind of situations?

fauxpark, 2019-04-01:
Which MCU are you trying to use, and what is the exact error(s) you're getting?

The pin type should be whatever the datasheet says. For example on the [ATmega32U4](http://ww1.microchip.com/downloads/en/devicedoc/atmel-7766-8-bit-avr-atmega16u4-32u4_datasheet.pdf#G1.998466), the GPIO pin descriptions specify that they are bidirectional, and that on reset they become tri-stated. In contrast, for the [AT89C2051](https://ww1.microchip.com/downloads/en/DeviceDoc/doc0368.pdf#G1050374), the description only says the pins are bidirectional.

It looks like the default ERC configuration produces a warning for a Power Output pin connected to a Bidirectional pin, and an error for one connected to a Tri-State pin. @poeschlr @evanshultz this is interesting as I remembered there are some AVR devices with AREF and GPIO sharing a pin eg. ATxmega32C3. Should these instead be passive then?

poeschlr, 2019-04-01:
MCUs are one of these devices where a generic symbol simply does not cut it. It is intended that the user copies the MCU symbol into the project and edit the pin types and pin names to fit the application. (The software defines what a pin really is. This hardware config should then be reflected by the symbol in KiCad. There is no way we can help here with the current file format limitations. With a bit of luck we get user selectable pin function in the new file format.)

rs-development, 2019-04-01:
@fauxpark This is exactly the same issue here. AREF Pin is a standart GPIO Pin, MCU is ATSAME51.
I am fine with it as it is, i just wanted to make sure you guys are aware of the problem

myfreescalewebpage, 2019-04-04:
@rs-development I think the response of @poeschlr answer you question isn't it ? Is it possible to close this issue ?

rs-development, 2019-04-05:
@myfreescalewebpage Question is answered with it.

closed 2019-04-05 by rs-development