KiCad/kicad-symbols #1717: CY7C65215 pin error?

magnusjjj, 2019-04-07:
Hi.

While manually copying a schematic from paper, i noticed that pin 20 on CY7C65215-32LTXI is listed as VDDD.  According to the datasheet (https://www.cypress.com/file/129956/download) under the pin list, the schematic I am copying, and the example in there, it should be VSSD. Am I missing something? New to kicad (or any eda).

myfreescalewebpage, 2019-04-07:
Hi @magnusjjj thanks for this issue, this is clearly a bug ! I propose myself to make the fix, since you are new to KiCad, this symbol has also several KLC violations. At the moment, you can connect pin 20 to VSSD according to me.
Cheers,
Joel

magnusjjj, 2019-04-07:
Sounds good to me ;).

myfreescalewebpage, 2019-04-07:
Pull request made.

myfreescalewebpage, 2019-04-09:
Fixed in #1726

closed 2019-04-09 by myfreescalewebpage