KiCad/kicad-symbols #1722: Audio jack upside down

clacktronics, 2019-04-08:
Perhaps this is a bit picky but It seems a bit awkward to have the shield connection at the top as this is typically a ground connection and conflicts with the convention of having ground marked downwards. Would it be welcomed to flip the jacks vertically or would this be a conflict with legacy libraries?   
[image]
[image]

evanshultz, 2019-04-08:
This is a fair point, but it would break existing schematics so we won't do it now. I'm marked this issue as a milestone for v6's library release where we have more latitude to change things. At that time, it will get revisited and dispositioned.

cpresser, 2019-09-14:
This applies to all `AudioJack*`in the library. Thats 22 parts in total.
One simple workaround is to use mirror in the schematic editor. 
[image]