KiCad/kicad-symbols #1741: Making multiple unrelated contributions

calebreister, 2019-04-13:
I'm currently waiting for #1428 to be reviewed, but have since been putting multiple smaller additions (many of which are likely already KLC-compliant) into a project-local library in order to avoid "polluting" my pull request with numerous smaller changes. I haven't created any other branches, since the [contribution guidelines](http://kicad-pcb.org/libraries/contribute/) say not to. Is there a "correct" way to make multiple pull requests to this project?

poeschlr, 2019-04-14:
Where do the contribution guidelines say that you should not make branches. I read it as the oposite. See point 4:
> Create a new branch in your local repository. This ensures that any local changes are not mixed with the master copy of the library until they are checked and merged by a librarian.

And 8 clarifies that every pull request is connected to a branch:
> Create a pull request (PR) to merge your branch into the KiCad libraries. Include links to reference documents (e.g. datasheets) in the comments, in addition to a description of the changes you have made.

myfreescalewebpage, 2019-04-14:
@calebreister yes @poeschlr is right, you should create a branch for each pull request you want to make :)

calebreister, 2019-04-15:
Ok, I get it now. This is the part that was confusing me:

> 10. If further changes are needed, **do not create a new branch**. Simply make any further changes to the same branch, and push these changes upstream. They will be automatically added to the PR.

closed 2019-04-15 by calebreister