KiCad/kicad-symbols #1753: DS1267 in Potentiometer_Digital has pin 13 and 12 swapped

pyroesp, 2019-04-15:
Datasheet:
[image]  
KiCad symbol:
[image]  

Unfortunately I just saw that, so my board will have a bodge wire.

myfreescalewebpage, 2019-04-17:
@pyroesp thanks for this report and sorry for this bad news :( Are you able to make a pull request to fix it ? Else I can do it myself, for future users. Let me known.
Thanks,
Joel

pyroesp, 2019-04-17:
Hey Joel,  
I didn't have time until today to make the pull request.  
Hopefully I've done it right.  

This is an old digital pot I needed to use in an old project.  
I'll be able to add a wire to the board to fix it, but yeah, I don't think anyone is ever going to use that part nowadays (except me).

Anyways, please check out pull request #1759 .

myfreescalewebpage, 2019-04-18:
Pull request has been merged.

closed 2019-04-18 by myfreescalewebpage