KiCad/kicad-symbols #1754: Web page or github page lisitng all the symbols?

E3V3A, 2019-04-16:
It would be useful to have a clear pictorial list of all the different basic symbols.
I.e. a table with the symbol name and schematic symbol.

It would help saving some time having to look for alternative symbols with obscure names.

aewallin, 2019-04-16:
are there python scripts that can read and render (PNG?) symbols and footprints?

poeschlr, 2019-04-16:
I seem to remember there was a post on the forum where somebody made a website that could show kicad library assets.
I also seem to remember there was a post about a tool that could export a list of library assets with pictures to a pdf.

I can not find either of them right now. But you could make a post on the forum and ask if somebody has such thing lying around. (I think such a question is generally better served on the forum.)

https://forum.kicad.info/

closed 2019-04-16 by poeschlr