KiCad/kicad-symbols #1772: Raspberry Pi board

myfreescalewebpage, 2019-04-21:
A work a currently in progress to include symbol/footprints for Raspberry Pi board. A list with proposed naming is given below. Checked will be added when new symbols are defined.

- 26 GPIO
  - Raspberry Pi 1 A => `RaspberryPi-1A`
  - Raspberry Pi 1 B => `RaspberryPi-1B`
- 26 GPIO with `RUN, GND`
  - Raspberry Pi 1 A => `RaspberryPi-1A_Run`
  - Raspberry Pi 1 B => `RaspberryPi-1B_Run`
- 40 GPIO
  - Raspberry Pi 1 A+ => `RaspberryPi-1A+`
  - Raspberry Pi 1 B+ => `RaspberryPi-1B+`
  - Raspberry Pi 2 B => `RaspberryPi-2B`
  - Raspberry Pi 3 B => `RaspberryPi-3B`
  - Raspberry Pi 3 A+ => `RaspberryPi-3A+`
  - Raspberry Pi 3 B+ => `RaspberryPi-3B+`
  - Raspberry Pi Zero => `RaspberryPi-Zero`
  - Raspberry Pi Zero W => `RaspberryPi-ZeroW`
- 40 GPIO with `RUN, GND`
  - Raspberry Pi 1 A+ => `RaspberryPi-1A+_Run`
  - Raspberry Pi 1 B+ => `RaspberryPi-1B+_Run`
  - Raspberry Pi 2 B => `RaspberryPi-2B_Run`
  - Raspberry Pi 3 B => `RaspberryPi-3B_Run`
  - Raspberry Pi Zero => `RaspberryPi-Zero_Run`
  - Raspberry Pi Zero W => `RaspberryPi-ZeroW_Run`
- 40 GPIO with `PEN, RUN`
  - Raspberry Pi 3 A+ => `RaspberryPi-3A+_Pen_Run`
  - Raspberry Pi 3 B+ => `RaspberryPi-3B+_Pen_Run`
- 40 GPIO with `PEN, RUN` and POE
  - Raspberry Pi 3 B+ => `RaspberryPi-3B+_Pen_Run_PoE`
- 40 GPIO with `RUN, GND` and `TV, GND`
  - Raspberry Pi Zero => `RaspberryPi-Zero_Run_Tv`
  - Raspberry Pi Zero W => `RaspberryPi-ZeroW_Run_Tv`

The purpose of the current work is to allow this kind of mounting:

[image]

myfreescalewebpage, 2019-04-21:
Proposition of design:

[image]

Format of DCM file:

```
$CMP RaspberryPi-xxx
D Raspberry Pi xxx, x.x GHz [single, dual, quad] core, xxx MB RAM, singleboard computer
K raspberrypi gpio sbc singleboard computer
$ENDCMP
```

Format of LIB file:

```
#
# RaspberryPi-xxx
#
DEF RaspberryPi-xxx J 0 20 Y Y 1 F N
F0 "J" -600 1250 50 H V L CNN
F1 "RaspberryPi-xxx" 100 -1350 50 H V L CNN
F2 "" 100 0 50 H I C CNN
F3 "" 100 0 50 H I C CNN
$FPLIST
 Raspberry*Pi*
$ENDFPLIST
DRAW
S -600 1200 700 -1300 0 1 10 f
X 3V3 1 -100 1300 100 D 50 50 1 1 W
X PoE1 1 200 1300 100 D 50 50 1 1 w
X GPIO15/RXD 10 -700 800 100 R 50 50 1 1 B
X GPIO17 11 -700 500 100 R 50 50 1 1 B
X GPIO18/PWM0 12 -700 400 100 R 50 50 1 1 B
X GPIO27 13 -700 -700 100 R 50 50 1 1 B
X GND 14 0 -1400 100 U 50 50 1 1 P N
X GPIO22 15 -700 -200 100 R 50 50 1 1 B
X GPIO23 16 -700 -300 100 R 50 50 1 1 B
X 3V3 17 -100 1300 100 D 50 50 1 1 P N
X GPIO24 18 -700 -400 100 R 50 50 1 1 B
X MOSI0/GPIO10 19 800 -400 100 L 50 50 1 1 B
X 5V 2 -300 1300 100 D 50 50 1 1 W
X PoE0 2 100 1300 100 D 50 50 1 1 w
X GND 20 0 -1400 100 U 50 50 1 1 P N
X MISO0/GPIO9 21 800 -300 100 L 50 50 1 1 B
X GPIO25 22 -700 -500 100 R 50 50 1 1 B
X SCLK0/GPIO11 23 800 -500 100 L 50 50 1 1 B
X ~CE0~/GPIO8 24 800 -200 100 L 50 50 1 1 B
X GND 25 0 -1400 100 U 50 50 1 1 P N
X ~CE1~/GPIO7 26 800 -100 100 L 50 50 1 1 B
X ID_SD/GPIO0 27 800 900 100 L 50 50 1 1 B
X ID_SC/GPIO1 28 800 800 100 L 50 50 1 1 B
X GCLK1/GPIO5 29 800 200 100 L 50 50 1 1 B
X PoE3 3 400 1300 100 D 50 50 1 1 w
X SDA/GPIO2 3 800 600 100 L 50 50 1 1 B
X GND 30 0 -1400 100 U 50 50 1 1 P N
X GCLK2/GPIO6 31 800 100 100 L 50 50 1 1 B
X PWM0/GPIO12 32 800 -700 100 L 50 50 1 1 B
X PWM1/GPIO13 33 800 -800 100 L 50 50 1 1 B
X GND 34 0 -1400 100 U 50 50 1 1 P N
X GPIO19/MISO1 35 -700 200 100 R 50 50 1 1 B
X GPIO16 36 -700 600 100 R 50 50 1 1 B
X GPIO26 37 -700 -600 100 R 50 50 1 1 B
X GPIO20/MOSI1 38 -700 100 100 R 50 50 1 1 B
X GND 39 0 -1400 100 U 50 50 1 1 P N
X 5V 4 -300 1300 100 D 50 50 1 1 P N
X PoE2 4 300 1300 100 D 50 50 1 1 w
X GPIO21/SCLK1 40 -700 0 100 R 50 50 1 1 B
X SCL/GPIO3 5 800 500 100 L 50 50 1 1 B
X GND 6 0 -1400 100 U 50 50 1 1 W
X GCLK0/GPIO4 7 800 300 100 L 50 50 1 1 B
X GPIO14/TXD 8 -700 900 100 R 50 50 1 1 B
X GND 9 0 -1400 100 U 50 50 1 1 P N
X GND ~ 100 -1400 100 U 50 50 1 1 P N
X GND ~ 200 -1400 100 U 50 50 1 1 P N
X PEN ~ -700 -1000 100 R 50 50 1 1 I
X Run ~ -700 -900 100 R 50 50 1 1 I
X TV ~ 800 -1000 100 L 50 50 1 1 O
ENDDRAW
ENDDEF
```

jstjst, 2019-04-21:
Design Changes:
- 416b6512b1d05d6c780e4af21a1384562cbc461a
- 763f27b643251de57b959f505b83c6b27cfc291c

[image]

jstjst, 2019-04-21:
I also added a field for the RPi model into the Footprint filter, since every model will need a separate footprint (especially when it comes to 3d models): 049bdc611ba57acd515b12e7b421782f1914343d
@myfreescalewebpage I think the template is now ready to create the symbols

myfreescalewebpage, 2019-04-22:
@jstjst this looks great if you want to submit the first symbols :)

jstjst, 2019-04-30:
Design Changes:
- 22a638c5ce50e710865779165099fd7bd1522a81
- e8cc88255ee267d3317683d6e74c85dd2e09fae2
[image]

jstjst, 2019-04-30:
- [ ] RaspberryPi-1A
  - [ ] RaspberryPi-1B
- [ ] RaspberryPi-1A_Run
  - [ ] RaspberryPi-1B_Run
- [ ] RaspberryPi-1A+
  - [ ] RaspberryPi-1B+
  - [ ] RaspberryPi-2B
  - [ ] RaspberryPi-3B
  - [ ] RaspberryPi-3A+
  - [ ] RaspberryPi-3B+
  - [ ] RaspberryPi-Zero
  - [ ] RaspberryPi-ZeroW
- [ ] RaspberryPi-1A+_Run
  - [ ] RaspberryPi-1B+_Run
  - [ ] RaspberryPi-2B_Run
  - [ ] RaspberryPi-3B_Run
  - [ ] RaspberryPi-Zero_Run
  - [ ] RaspberryPi-ZeroW_Run
- [ ] RaspberryPi-3A+_Run_PEN
  - [ ] RaspberryPi-3B+_Run_PEN
- [ ] RaspberryPi-3B+_Run_PEN_PoE
- [ ] RaspberryPi-3B+_PoE
- [ ] RaspberryPi-Zero_Run_TV
  - [ ] RaspberryPi-ZeroW_Run_TV
- [ ] RaspberryPi-Zero_TV
  - [ ] RaspberryPi-ZeroW_TV

jstjst, 2019-04-30:
@myfreescalewebpage  I'm running in some trouble right now, because I want to create some symbols as an alias. But they need a different footprint and footprint filter, since there are different RPi models. But that is not possible. If possible, I want to be able to use aliases.
Any ideas?

myfreescalewebpage, 2019-05-01:
If the footprint is not the same, then they can't be alias. You can make the first one. Copy/paste. then change footprint of the copy.
Joel

jstjst, 2019-05-05:
@myfreescalewebpage I created the first set of symbols: #1820

jstjst, 2019-05-13:
40 GPIO symbols:
#1852 
    Raspberry Pi 1 A+ => RaspberryPi-1A+
    Raspberry Pi 1 B+ => RaspberryPi-1B+
    Raspberry Pi 2 B => RaspberryPi-2B
    Raspberry Pi 3 B => RaspberryPi-3B
    Raspberry Pi Zero => RaspberryPi-Zero
    Raspberry Pi Zero W => RaspberryPi-ZeroW
#1851 
    Raspberry Pi 3 A+ => RaspberryPi-3A+
    Raspberry Pi 3 B+ => RaspberryPi-3B+