KiCad/kicad-symbols #1802: Small question, why git/kicad modify thing that i don't want to

JonathSpirit, 2019-04-30:
I just modified only one thing (the position offset on the 74LS374), and kicad change the position in the file of the 74HC14 (same data in a different place) that i never touch :/
so what do i do ? I commit the change as well ?

[image]
[image]

My small change :(
[image]

evanshultz, 2019-04-30:
Yes, just commit it.

The devs have a bad habit (IMO) of changing the line ordering during save between various builds of KiCad. This is a common problem that we just deal with because we've seen it so many times.

JonathSpirit, 2019-04-30:
ok thanks !

poeschlr, 2019-05-02:
The best option would be to open the lib, save it without changing it (use the move command on a pin but place it at the same place to enable saving) check with git status if something changed. If it did make a separate commit with "saved with current stable" and then make your changes on a later commit. Makes it much easier for us maintainers to see what goes on. (We can then squash merge at the end to keep the master branch clean of such commits.)

closed 2019-05-02 by antoniovazquezblanco