KiCad/kicad-symbols #1809: Transformer left unconnected in schematics

Salamandar, 2019-05-02:
I'm trying to use transformers, but the schematics looks completely lost. 
You can see on the screenshot that all components that **should** be connected to the transformer have an error "unconnected pin".
Obviously, the pins are still unconnected on the PCB.
I tried with different transformer symbols, same issue.

[image]

evanshultz, 2019-05-02:
@Salamandar 
What symbol are you using? The official library doesn't have a symbol named `Transformer_1P_1S`.

If this a symbol you made yourself, you really aren't in the right place to ask about this. This repo is regarding the official KiCad library. I would suggest https://forum.kicad.info/ for "using KiCad" questions and issues.

But anyway, my first thought is to check that the pins are on a coarse grid. Otherwise it can be hard to connect wires to pins. Our library uses a 100mil pin grid for this reason.

Salamandar, 2019-05-02:
I'm using the Device:Transformer_1P_1S, in the official libraries.w
Found it here in the repo : https://github.com/KiCad/kicad-symbols/blob/844bc98d2a6f57f80e920b28181ab4bf68b5b39f/Device.lib#L11376

But, actually, I found that the bug is not in the libraries but in Kicad itself. When updating (sch -> PCB), the new components are not connected to the already-existing on the pcb.
I'll report it to the kicad issues.

DanSGiesbrecht, 2019-11-02:
@Salamandar why does your schematic have pins numbered `1,7,13,9`? The library symbol has `1,2,3,4`. Can this issue be closed?

Salamandar, 2019-11-05:
The library was updated (by my distro), and the issue disappeared. I guess there was a bug in the libraries at the time, but already fixed in master.

closed 2019-11-05 by Salamandar