KiCad/kicad-symbols #1875: NXP Lin Transceiver TJA1021 RX/TX swapped

Deadeye5589, 2019-05-22:
The pin numbers and naming of the NXP Lin Transceivers TJA1021 from the Interface_CAN_LIN library do not match the datasheet. Rx and Tx are swapped in comparison to the datasheet. 
This might be related to the fact the the NXP chipsets are not following to the regular convention of Rx<>Tx. For the NXP Lin and Can chipsets the UART connection must be Rx<>Rx and Tx<>Tx. This is clearly stated in the datasheet and also shown the the application circuit example within the datasheet. Therefore I like to request to correct the NPX symbols to match the datasheet. The fact that the pins in the symbol are swapped leads to failures in the schematic creation when following the datasheet without checking the symbol for correct pining. Which I think might be the case for most of the users.

evanshultz, 2019-05-23:
Thanks for noticing this and taking the time to write it up.

Datasheet: https://www.nxp.com/docs/en/data-sheet/TJA1021.pdf?

Current symbol:
[image]

One explanatory section of the datasheet is here:
[image]

I've not used these parts, but it seems you're correct. There are two fixes (for both symbols):
1. Swap pins numbers for pins 1 and 4.
1. Change the type of the RXD pin to Open Collector (KiCad doesn't have Open Drain).

@Deadeye5589 
Would that fix the issue you noted? If so, would you be willing to submit a PR to correct these symbols?

Deadeye5589, 2019-05-23:
Yes swap pin 1 and 4 would definetly fix this issue for this component. But I checked some of the other NXP transeivers as well and they have the same problem. Therefore I would suggest, that I check also the remaining NXP components and start a PR for the whole bib.

Deadeye5589, 2019-05-24:
Fix commited and ready for review

evanshultz, 2019-05-28:
https://github.com/KiCad/kicad-symbols/pull/1880#issuecomment-495998821

Deadeye5589, 2019-06-01:
Bugfix merged. 
Topic closed

closed 2019-06-01 by Deadeye5589