KiCad/kicad-symbols #1882: Symbol MCP6004 has wrong component name: L2902

wmzart, 2019-05-26:
See the following screenshot:

[image]

antoniovazquezblanco, 2019-05-27:
MCP6004 is an ALIAS of LM2902. This means that all the drawing and pin information is taken from LM2902 while datasheet, description and keywords are overriden by the values of MCP6004.

You can check that information in the library symbol properties dialog in the aliases tab.

I hope it helps!

antoniovazquezblanco, 2019-05-27:
By the way, in this case, there is no issue with the library but if you wish to change the behaviour of KiCad so that it renders the aliases information you should report that at https://bugs.launchpad.net/kicad

Thank you!

closed 2019-05-27 by antoniovazquezblanco

poeschlr, 2019-05-27:
My guess is that this would be classified as a new feature and would therefore not happen for v5. It makes little sense to implement anything alias related for v6 as the whole alias stuff will be dropped in favor of the inheritance stuff. (Or possibly also the new atomic stuff as suggested by seth. I am not yet convinced that we should use both the inheritance stuff and the atomic stuff)

wmzart, 2019-05-27:
Thanks for pointing out the alias feature. In that case I will not file a bug report and wait for what is going to be implemented in v6.