KiCad/kicad-symbols #1899: Add LF442 and LM1458 as aliases of the LM2904

ZirconiumX, 2019-06-02:
As far as I can tell, the [LF442](https://www.ti.com/lit/ds/snosd59/snosd59.pdf) and [LM1458/LM1558](https://www.ti.com/lit/ds/symlink/lm1458.pdf) have the same pin out as the [LM2904](https://www.ti.com/lit/ds/symlink/lm2904.pdf). I could file a PR if you want me to, but I don't think it's worth the effort of CLA-signing etc for a one-line change.

fauxpark, 2019-06-04:
This would not be a one line change as you would also need to add entries in the .dcm file.

evanshultz, 2019-06-20:
@ZirconiumX 
We don't take requests so I'm going to close this. You were given a great response above that explains why it's a bit more involved. If you would like to file a PR to add the parts that would be great.

closed 2019-06-20 by evanshultz