KiCad/kicad-symbols #1910: APA102 Footprint incorrect

wirehead, 2019-06-09:
I printed out a PCB that used the APA102 symbol provided in the built-in library and was validating all of the footprints against the printout.

Pictured, a bit hard to see, is an APA102 using the library's footprint. You can see that the spacing on the pads doesn't match.

[image]

[The SparkFun library](https://github.com/sparkfun/SparkFun-KiCad-Libraries) seems to be correct when I checked it physically against the part.

poeschlr, 2019-06-09:
please post a link to the datasheet

wirehead, 2019-06-09:
That's the APA102-2020 you linked to.  This concerns the APA102 which uses the LED_RGB_PLCC-6 footprint.

Here's a datasheet: http://www.led-color.com/upload/201506/APA102%20LED.pdf

poeschlr, 2019-06-10:
Your datasheet does not contain a dimensioned drawing. It is therefore useless to determine which footprint should be used

evanshultz, 2019-06-10:
Perhaps I can add a bit more info to clarify...

The APA102-2020 symbol references an atomic and, I hope, correct footprint:
[image]

But the APA102 symbol uses a more generic footprint:
[image]

The product page for APA102 appears to be at http://szledcolor.com/productshow.asp?id=878&sid=199, but it is the datasheet you linked above without dimensions. It appears you have found an error in the library but we can't take any action to correct it right now because we're missing the information to tell us what the correct footprint should be.

If you can identify the correct package and/or footprint information, would you be willing to submit a footprint for this part and also update the symbol to use that footprint?

fauxpark, 2019-06-11:
Per #1501 it seems the footprints for both are correct. The submitter links to an Adafruit datasheet, containing the same drawing, and says it is PLCC-6 (5050) - which is exactly what the part in the OP here is.

I think the main problem is that while the 2020 symbol has the `-2020` suffix, the 5050 symbol has no suffix at all.

wirehead, 2019-06-11:
No, unfortunately, I think you underestimate just how annoying the modern SMT LED industry has become.  Like, sure, fiat lux, but can we fiat lux in a discrete number of packages instead of ten zillion almost-but-not-quite-identical packages, none of them with standardized descriptive names?

So, I found this discussion from the forums: https://forum.kicad.info/t/smd-rgb-led-footprint-5050/653/8 and the associated PR from two years ago.

I just did my check and if the APA102 uses the `LED_SMD:LED_RGB_5050-6` footprint instead of the `LED_SMD:LED_RGB_PLCC-6` footprint, it fits.  So I think this problem is easily solved just by changing footprints.

Also, yeah, it's probably proper and least-surprise to call it a `APA102-5050` as compared to a `APA102-2020`.

fauxpark, 2019-06-11:
Ah, I was under the impression the PLCC-6 footprint was 5050... it is actually 3434.

poeschlr, 2019-06-11:
The APA102-2020 is called that by the manufacturer this is why this name is used. The datasheets linked here call the part APA102C which would be the proper name of it.

evanshultz, 2019-06-11:
@wirehead 
If you can submit a PR to change the symbol to use a correct footprint, since things are broken, we will accept that right away.

But I still haven't seen a datasheet that shows dimensions so we can verify that the `LED_RGB_5050-6` footprint is correct. Are you able to find some documentation?

evanshultz, 2019-06-20:
@wirehead 
It seems like the issue is sorted out above. Are you willing to make a PR to fix this? Would be nice to squeeze it into 5.1.3 which is coming soon.

evanshultz, 2019-07-11:
I contacted the vendor to request a footprint or an adequately-dimensioned package drawing for APA102.

evanshultz, 2019-07-25:
No answer after multiple inquires. I'm giving up.

Is the expectation that changing the footprint for this symbol to `https://github.com/KiCad/kicad-footprints/blob/master/LED_SMD.pretty/LED_RGB_5050-6.kicad_mod` will fix the issue?

wirehead, 2019-07-28:
PR submitted.  Very sorry for going silent on this.

PR #2032 (2019-07-29): Change footprint for the APA102 5050
Fixes #1910

 - change footprint from LED_SMD:LED_RGB_PLCC-6 to LED_SMD:LED_RGB_5050-6
 - see KiCad/kicad-symbols#1910 for discussion.

Since we don't have a datasheet and nobody from the manufacturer wants to give any more details, here's a board I fabbed at DirtyPCB's from a file generated in KiCad using the LED_RGB_5050-6 footprint with a real live APA102 part I ordered from DigiKey.

*sigh*
closed 2019-07-29 by evanshultz