KiCad/kicad-symbols #1926: Missing symbol for 74HC165A (serial in parallel out w latch) and associated libraries

rjj100, 2019-06-18:
Missing symbol for 74HC165A (serial in parallel out w latch) and associated libraries as of 6 - 17-19

poeschlr, 2019-06-18:
We are sorry but we can not take requests for additions. The library does not claim to have a symbol for every component. The issue tracker is for reporting bugs or discussions about the library convention. (The later mainly such that we maintainers can get input from users.)

In general making a symbol is quite easy and a basic thing you need to learn anyways. Take a look at https://forum.kicad.info/t/tutorial-how-to-make-a-symbol/13336/
If you get stuck ask for help over on the forum.

closed 2019-06-18 by poeschlr

rjj100, 2019-06-18:
I understand, thanks. I’ve only found KiCAD 1 day ago and will be awhile before I can contribute.