KiCad/kicad-symbols #1951: Symbol design for operational transconductance amplifier (OTA)

lorem-ipsum, 2019-07-05:
I want to make a symbol for the [OPA861](https://www.ti.com/lit/ds/symlink/opa861.pdf) OTA, but am not sure on what symbol to use. The library has already the `CA3080(A)` and the `LM1360`/`LM1370`. They use the design of a regular op-amp, adding the bias at some location. As these are not regular op-amps I think this should also made clear by the symbol.

The datasheet lists some symbols on page 12, but those only have the input and output pins, lacking the biasing pins. The symbol at the [corresponding Wikipedia article](https://en.wikipedia.org/wiki/Operational_transconductance_amplifier) seems to look good, but I am not sure whether this follows some standard.

Do you have any suggestions / preferences? I would like to add the mentioned OPA860, and would also update the existing symbols to have a consistent style throughout all OTAs.

evanshultz, 2019-07-05:
First, a disclaimer: I designed the existing symbols, so I'm probably biased.

Cool part. Thanks for asking about it.

What is unclear about the existing triangular symbols? IMO common OTAs are well-known by those who use them so seeing the part number on the schematic is enough. And the BIAS pin provides uniqueness compared with the normal opamp having only two input pins.

CA3080 is a single symbol, but LM13x00 is broken up with the transistors in separate units. I like this style because it gives flexibility in the schematic layout. And for multi-unit triangle and logic symbols the power pins should be in a separate unit. What did you have in mind for OPA861's symbol, since I can image symbol designs that looks unlike either of those?

One thing to keep in mind is that existing symbols should not have pin locations moved to avoid breaking schematics. Unless it fixes an error, we'll only do that at major KiCad version releases.

lorem-ipsum, 2019-07-05:
You're probably right about the understanding based on the bias pin. I mostly stumbled about the CA3080 which has the bias not between the inputs, but on the side (like a supply).

I also was wondering whether one should include the diodes and circles in the housing, but thats not doable without moving pins around. So I will stick to the simple design of the LM13x00.